FreeCAD/FreeCAD #12819: [Problem] Cannot use EasyAlias because Spreadsheet cell names start with `'`

TheRealOne78, 2024-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I tried using EasyAlias to alias my spearsheet cells, but I can't just do that because for whatever reason newer versions of FreeCAD put a `'` at the start of a string that cannot be removed. See first image at 'Content'.
[image]

This results into EasyAlias failing to alias a cell, because the string starts with an invalid character:
[image]

### Full version info

```shell
[code]
OS: Gentoo Linux (Xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.2.Unknown
Build type: RelWithDebInfo
Python 3.11.7, Qt 5.15.12, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.0
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
  * ProDarkThemePreferencePack 1.0.0
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-09:
@TheRealOne78 what is EasyAlias? Is this issue related to a third party addon or can be reproduced with the Spreadsheet WB only?

TheRealOne78, 2024-03-09:
EasyAlias is a macro addon (also listed on [freecad wiki](https://wiki.freecad.org/Macro_EasyAlias)), where you select a cell (or more, but needs to be the same column) and then aliases the cell next to it. I'd say it's a good approach to label your data in a spreadsheet

[image]

But doing this manually is too much work, so here should come EasyAlias to help.

EasyAlias's behaviour is to query copy the contents in the selected cell and then alias the cell next to it.

But for a while (maybe 4 months?), I've not been able to use EasyAlias due to this weird change (FreeCAD puts a quote `'` in front of each non-expression cell, which was not the case about 4 months ago)

JohnOCFII, 2024-03-09:
> EasyAlias is a macro addon (also listed on [freecad wiki](https://wiki.freecad.org/Macro_EasyAlias)), where you select a cell (or more, but needs to be the same column) and then aliases the cell next to it. I'd say it's a good approach to label your data in a spreadsheet

EasyAlias is a great tool.  I use it almost daily.  Shortly after FreeCAD added the tick mark to each cell, the author of EasyAlias updated his tool to work.  Are you sure you are using the latest version?  You can check in the AddOn Manager: 

[image]

TheRealOne78, 2024-03-09:
> Shortly after FreeCAD added the tick mark to each cell, the author of EasyAlias updated his tool to work.

FreeCAD doesn't add a tick mark '\`', but rather a single quote mark `'`.
<br>

> Are you sure you are using the latest version?

Yes, I've tried removing and reinstalling the addon too. It's currently at the latest version '2023.04.27'
[image]

luzpaz, 2024-03-09:
CC @mwganson

TheRealOne78, 2024-03-09:
> CC @mwganson

I mean.. sure.. I will do that.... but why is this supposed to have a single quotation mark for every non-expression cell in the first place?

furgo16, 2024-03-09:
Related https://forum.freecad.org/viewtopic.php?p=706749, but it seems it was already resolved a while back.

mwganson, 2024-03-09:
The issue should have already been resolved.  Check line 71 of the macro code:

```
REPLACEMENTS = {
    " ": "_",
    ".": "_",
    "ä": "ae",
    "ö": "oe",
    "ü": "ue",
    "Ä": "Ae",
    "Ö": "Oe",
    "Ü": "Ue",
    "ß": "ss",
    "'": ""
```

Line 71 is the last line above.

This is where the single quote character gets removed before attempting to create the alias.

maxwxyz, 2024-03-10:
I'll close this then. @TheRealOne78 please use the latest version and comment if there are still issues.

closed 2024-03-10 by maxwxyz

TheRealOne78, 2024-03-10:
ok that's odd.. I am using the latest version of this plugin yet FreeCAD still doesn't have this change:
[image]

[image]

I guess I could add it manually..

But the question still remains:
> but why is this supposed to have a single quotation mark for every non-expression cell in the first place?