FreeCAD/FreeCAD #12820: [Problem] Use only selected parts of a sketch / Features with different profiles of a master sketch

maxwxyz, 2024-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently it is not possible to create a *master sketch* with multiple, overlapping islands of wires and use the individual parts of the sketch in the 3D view for features. Always the entire sketch has to be used to create features. If there are open loops / wires or not one single closed profile, most PartDesign features fail.
The idea is to create a sketch and use a a single loop / closed profile or combinations of the enclosed sketch for features in FreeCAD, for example PartDesign WB (Pad, Pocket, Revolve,...), Part WB (Extrude,...) or elsewhere. 

In other CAD software, when activating a sketch based feature (e.g. Pad) the user can select the parts of the sketch which should be included in the profile to be extruded. When hovering the sketch, the different - closed - profiles of overlapping sketch geometry are highlighted and can be selected individually to be added or removed from the final group of profiles which will be used for that feature.

Example from Fusion360:
[image]

Example from Onshape:

[image]
[image]

This also lets the user create a single sketch (sometimes referred to as *master sketch*) and referencing different parts for different features, without the need to copy the sketch, redraw the sketch, recreate dimensions or create external geometry references and redrawing or constraining with regard to them.

This was also mentioned as issue at the FreeCAD Day 2024 Complaint Session:
**Unable to easily apply operations to selected parts of a sketch**

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

PartDesign

### Anything else?

Maybe related:
https://github.com/FreeCAD/FreeCAD/issues/10006
https://github.com/FreeCAD/FreeCAD/issues/5717

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andyylin, 2024-05-30:
Yes, this is an amazing feature since using external geometries is cumbersome and the links often break when sketches are modified.

nazar-pc, 2024-07-30:
I believe this works now in dev branch in PartDesign as shown in [MagoJelly's video](https://www.youtube.com/watch?v=IjzhUCl3gXg) and discussed extensively in https://forum.freecad.org/viewtopic.php?t=89381 (even though discussion started with a slightly different reason).

It doesn't work this way in Part workbench though (not yet at least) and I'd really like to see that changing.

maxwxyz, 2024-11-09:
It works by manually selecting individual edges which form a closed loop. I'd consider this a workaround.
Ideally the closed loops / islands are highlighted like a face to choose them.

trilbytim, 2024-11-09:
@maxwxyz I strongly agree with you! Yes, selecting the edges is clumsy, the Real Thunder branch version (see my comment above) with automatically generating faces is MUCH nicer to use. It would be great to see the Real Thunder method in the main branch.

FEA-eng, 2025-01-03:
> I've been using the Realthunder branch for some time which has a 'Make Internals' feature and I find this functionality SOOO useful!

What's important to mention, the `Make Internals` property is already in FreeCAD 1.0 but doesn't work. Thus, it's not only a feature request but also a bug report at this point. The whole "multiple pads from one sketch" thing was implemented unintentionally as part of the Realthunder's TNP code. But now it needs some work to operate properly. Currently, only edges can be selected, there's no splitting (`Make Internals`) and subsequent coloring of the profiles.

greenscreenflicker, 2025-01-24:
Considering bow many duplicates that issue has, a Feature would bring value to plenty of users.

techy-robot, 2025-06-28:
I almost opened a bug report for this lol, because the Make Internals option was there, but I had no way to actually select or see sketch sub faces for part design.

theosib, 2026-01-14:
I'd like to take a crack at this, so I thought I'd share what I found and get some feedback.

It looks like the first thing I need to do is modify Part::Extrusion to use PropertyLinkSub Base instead of PropertyLink Base.

I made that change locally and tested it out. Backward compatibility is preserved (extrude.Base = sketch still works). Face selection from any Part::Feature works (extrude.Base = (box, ['Face1'])). And all 127 existing Part tests pass.

Where I'm at:
The faces created by MakeInternals are stored in the sketch's InternalShape property, not the main Shape. The Feature::getTopoShape() function doesn't include these internal elements, so selecting them (e.g., extrude.Base = (sketch, ['Face1'])) fails to find the sub-shape.

So the questions are:
Is there an existing pattern for accessing a sketch's internal faces from Part workbench? Should getTopoShape() be extended to handle this, or should Part::Extrusion handle sketches specially?

Thanks!