FreeCAD/FreeCAD #12821: [Feature] Implement a preference to disable the gray out of invisible items in the tree

NomAnor, 2024-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@adrianinsaval in the forum suggested that the gray out of invisible items in the tree could be disabled separately to the visibility icon.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36330 +9 (Git)
Build type: Debug
Branch: develop
Hash: cfcc98def95828dd1059a2faa2f9a3da4d32cd94
Python 3.11.8, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.3.0
Locale: English/United States (en_US)
Installed mods: 
  * Manipulator 1.5.7
  * ThreadProfile 1.84.0
  * DynamicData 2.60.0
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.0.0
  * Assembly3 0.11.3
  * QuickMeasure 2022.10.28
  * fasteners 0.4.21
```

### Subproject(s) affected?

None

### Anything else?

Currently the gray out is disabled when the visibility icon is enabled. This should be an easy change, if there is no strong oppisition for another preference, I will create a PR.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-03-09:
Is there a reason to disable the grey out at all?

maxwxyz, 2024-03-09:
Do we really need another setting for such a thing or could we define the default behavior properly.

NomAnor, 2024-03-09:
Sorry, here is the forum thread: https://forum.freecad.org/viewtopic.php?t=85108

@adrianinsaval In the end it's personal preference. I disabled it because it was disabled in LinkMerge.

Here is a comparision:
[image]

I'm fine with removing the disable, and keep the original gray-out even if the icon is enabled.

yorikvanhavre, 2024-03-11:
Indeed when people use the icon, it doesn't make so much sense to have the name greyed out. But it could go together: Do you want the icon or the greyed text, basically.

NomAnor, 2024-05-28:
@obelisk79 any opinion from the DWG?

If there is no clear preference, we could close the issue and keep it as is.

NomAnor, 2024-05-28:
Closed for #14342 which discusses more general UX for the icon

closed 2024-05-28 by NomAnor