FreeCAD/FreeCAD #12822: [Feature] Allow toggling the transparency with the visibility icon

NomAnor, 2024-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@Pendelf and @adrianinsaval suggested in the forum, that the transparency could also be toggled with the visibility icon.

Two workflows were suggested:
1. Alt + Left click on the icon
2. Middle click on the icon

A tooltip explaining this should also be added.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36330 +9 (Git)
Build type: Debug
Branch: develop
Hash: cfcc98def95828dd1059a2faa2f9a3da4d32cd94
Python 3.11.8, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.3.0
Locale: English/United States (en_US)
Installed mods: 
  * Manipulator 1.5.7
  * ThreadProfile 1.84.0
  * DynamicData 2.60.0
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.0.0
  * Assembly3 0.11.3
  * QuickMeasure 2022.10.28
  * fasteners 0.4.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-09:
How is an active reduced transparency displayed?
What would happen if you click on sketches, sub-features, spreadsheets, bodies of parts, ...?
Could you please link the forum discussion as well, TIA.

NomAnor, 2024-03-09:
Sorry, here is the forum thread: https://forum.freecad.org/viewtopic.php?t=85108

In #12298 @obelisk79:
> I had also suggested that visibility toggles be handled as a tri-state toggle. Visible, semi-transparent, hidden. However I expect such a change in functionality to be somewhat controversial unless it also was made optional.

We need different icons for the different modes (visible, transparent, hidden).

One implementation would be, if the transparency if toggled with a different action (e.g. middle click), a left click on a transparent iitem makes is hidden and then visible. A middle click would toggle between transparent and visible. 

It seems that #11353 shows a problem with some items that should or can't be transparent. This feature would probably need some kind of function to tell the code that a DocumentItem can be transparent. The visibility icon is also shown for all objects, event when they can't really be invisible (lacking a representation in 3D). This could also be fixed by such function.

obelisk79, 2024-03-10:
Alt-click or middle click may end up being the right answer for the control of this, however it would suffer from poor discoverability generally, even with a verbose tooltip.

While a 3 state cycle using click/spacebar would not have that problem. The issue that is difficult to determine is whether or not a simple tristate toggle is a notable inconvenience in actual use or not.