FreeCAD/FreeCAD #12823: Path - Tool Library

boboxx, 2024-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm in the process of adding some tools to the path tool library. I have created a simple end mill. If i go back to edit the endmill after it's been saved the digits in my name gets incremented/changed.

In the case I saved the tool as: 2F-D6.35-L31.75, but If I open up my tool again it's now showing 2F-D6.35-L31.076
[image]

### Full version info

```shell
OS: macOS 14.3.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```

### Subproject(s) affected?

Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-03-09:
Can you zip and drag drop (here in to github textbox) an example file for testers to reproduce ?

boboxx, 2024-03-09:
Here is the file and the end mill
[sample.zip](https://github.com/FreeCAD/FreeCAD/files/14546806/sample.zip)

PR #21425 (2025-06-02): CAM: Integration of the tool and shape editors from BTL (Better Tool Library)
This very large PR essentially replaces CAM Tool Management completely, as part of the [Better Tool Library](https://github.com/knipknap/better-tool-library/) integration.

## Relevant Issues

Discussion: #9466

I also went through all open CAM issues to find any that may be resolved or otherwise affected:

- Resolves: #8201
- Resolves: #8526
- Resolves: #9625
- Resolves: #11522
- Resolves: #12712
- Likely resolves: #12823, but should be tested to be sure
- Possibly resolves #13603, to be tested
- Resolves: #17083
-  [... cut]