FreeCAD/FreeCAD #12824: [Problem] Render quality of the 3D view / Appearance of geometry and environment

maxwxyz, 2024-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the FreeCAD Day 2024 Complaint Session:
**Lack of ambient inclusion shading makes it difficult to perceive orientation and details**
The lack of ambient occlusion shading or the appearance / rendering in the 3D view makes it difficult to perceive orientation and details depending on complexity and person.
This includes material display, lightning options (shadows, multiple lights, HDRI, environment illumination), and display of the geometry. There is no way to display textures.
Example:
Currently there is one light in FreeCAD, with the default setting it is always head on in the direction of the camera. Shiny objects appear totally white, whereas if you rotate the light there will be areas which are totally black. 
Demo: https://github.com/FreeCAD/FreeCAD/issues/11166#issuecomment-1828460603

Another point that came up which would be fixed with a better render in the 3D view:
**Want a simple button for a quick lofi render**
Renders are difficult but I just want to have a vague image to show someone. I tend to sit there turning lines off and changing the background and screenshoting because that's easier than making sort of a hard time

Related issues to address appearance problems:

- https://github.com/FreeCAD/FreeCAD/issues/11166
- https://github.com/FreeCAD/FreeCAD/issues/10614
- https://github.com/FreeCAD/FreeCAD/issues/9636
- https://github.com/FreeCAD/FreeCAD/issues/10904
- https://github.com/FreeCAD/FreeCAD/issues/10905
- https://github.com/FreeCAD/FreeCAD/issues/10210
- https://github.com/FreeCAD/FreeCAD/issues/5570

In @realthunder Link branch there is currently an **experimental renderer** included, which fixes most of the problems, (including the closed section view https://github.com/FreeCAD/FreeCAD/issues/5570#issuecomment-1162811460).

Example of the implementation in link branch:
[image]
[image]
[image]
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-04-07:
Here's a comparison using [this model](https://forum.freecad.org/viewtopic.php?t=63740) and default rendering settings in each software (apart from adjusted light orientation):

FreeCAD:

[image]

FreeCAD Link:

[image]

SolidWorks:

[image]

Fusion 360:

[image]

obelisk79, 2024-07-03:
I firmly believe, and have been bringing this up every time this comes along, that FreeCAD's 3d shaded views need to include an optimized ambient occlusion shader. Fusion360 has probably the best implementation of this listed above.

OnShape does the same and uses OpenGL ES and renders this with excellent performance within a web browser.

[image]

maxwxyz, 2024-10-24:
As demonstrated here https://www.youtube.com/watch?v=E75hptWOSZA a three light setup can also be achieved with the Arch Textures addon

marcdus, 2024-12-06:
I am trying to move from onshape to freeCAD and noticed the essential omission of **global illumination**. Since we are not used to looking at objects in outer space (where indeed there is no global illumination) we expect even the shaded sides of an object to be lit by a certain amount. This can sort of be simulated by adding emissive light to objects (see pictures) but of course that is a hack and has other adverse effects.

Adding global illumination would also remove the need for X-point light sources as onshape's view seems to do just fine with one directional light. The Link branch looks really nice but I think simple lighting with no self-shadowing is preferable since the shadows again can lead to poor visibility.

I have just started reading the freeCAD github and only checked 5-10 issues so far, let me know if I missed this point in previous discussions.

[image]
_Onshape (left) has a more usable contrast between directly lit and shaded areas._

[image]
_Emmissive object properties help, sort of simulating global illumination. However, it of course affects the general look as normal materials clearly shouldn't glow._

yorikvanhavre, 2024-12-06:
My 2 cents here:

- direct shadows are immediately doable (Link branch + several other tests done in the past), everything for it in coin already. An interesting preliminary step would be to compare these solutions and set up a plan for discussion
- One still need to define how a "virtual ground plane" should be done
- Global illumination is probably out of reach for now. Too complex scene redesign and heavy calculations which are not our speciality
- Ambient occlusion might be doable with GLSL shaders. This would require more research. If it turns out yes, that might be not such a far goal...

marcdus, 2024-12-06:
You raise some good points @yorikvanhavre and thanks for explaining. I think my terminology was incorrect before. I meant to refer to basic ambient lighting, simply raising the minimum brightness of every face uniformly. On wikipedia it is referred to as [ambient lighting](https://en.wikipedia.org/wiki/Shading#Ambient_lighting). I mentioned it because in contrast to ambient occlusion and shadow rendering it should require no additional computation and be simple to implement.

marcdus, 2024-12-06:
To illustrate why I don't think shadowing will improve visibility:
- Blender render with black as ambient color, no shadows:
  [image]
- Same but with shadows:
  [image]

Notice how the shadows don't help. What really helps is some amount of ambient light, and a material-penetrating directional light, as is also the case in the default blender viewport:
[image]

obelisk79, 2024-12-06:
Perhaps a 3 point lighting scheme as seen here could serve well? I've not explored this kind of setup though. There's may be drawbacks. 

https://youtu.be/E75hptWOSZA

yorikvanhavre, 2024-12-10:
In any case, being able to configure the default scene lighting better (backlight and/or ambient light, for ex) would be an improvement too.

mcolinp, 2026-04-27:
I would really love to see an option to use an HDR for the viewport lighting. This would bring much improved realism, especially for glossy surfaces that need reflections to truly look accurate. Most programs offer the ability to use an HDR image for lighting, even in combination with the standard lights.