FreeCAD/FreeCAD #12825: Readme not showing properly for add-ons

APEbbers, 2024-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For my workbenches (TitleBlockWB and Bill of Materials WB) the readme is not shown.
I only see a few list bullits.

I compared my readme's with others, but I couldn't find obvious differences.
This goes for FreeCAD 0.22dev. on FreeCAD 0.21, the actual webpage is shown.

### Full version info

```shell
OS: Windows 11 build 22635
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36121 (Git)
Build type: Release
Branch: (HEAD detached from c7ac9ee848)
Hash: 4932f17dc57eec8a91767add5c21a760138a7ac8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Netherlands (en_NL)
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.64
  * Assembly3 0.12.2
  * Assembly4 0.50.9
  * Behave-Dark-Colors 0.1.1
  * BillOfMaterials 0.0.5.10
  * boltsfc 2022.11.5
  * CfdOF 1.25.2
  * CommandPanel
  * Design456 0.0.1
  * Dracula 0.0.5
  * EM
  * ExtremeProDark 2.9.0
  * fasteners 0.5.13
  * FEMbyGEN 2.1.0
  * FEM_FrontISTR 0.1.0
  * frame 0.1.1
  * Freecad-Built-in-themes-beta 1.2.2
  * freecad.gears 1.2.0
  * IconThemes
  * Launcher
  * NikraDAP 2.0.0
  * OpenDark 2023.12.17
  * pcb 6.2023.1
  * Plot 2022.4.17
  * ProDarkThemePreferencePack 1.0.0
  * QuickMeasure 2022.10.28
  * Render 2024.1.23
  * SearchBar 1.0.2
  * sheetmetal 0.4.2
  * Ship 2022.4.11
  * TabBar
  * TitleBlock 0.5.2
  * ToolbarStyle
  * WorkFeature-WB
```

### Subproject(s) affected?

Addon Manager

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-09:
I can reproduce, it looks the same, but only on certain addons. The readme on GitHub looks normal
@chennes FYI

chennes, 2024-03-11:
The problem is that your markdown contains HTML, and the Qt parser doesn't understand it. I'll write some code to strip it out, but you will probably want to modify your README to replace those tags.

PR #12857 (2024-03-11): Addon Manager: Strip some HTML tags from Markdown
Fixes #12825. Even with "GitHub flavored Markdown" enabled, Qt's Markdown processor chokes on some HTML tags. This small method removes the tags that are known to break rendering. It may be extended in the future to do a more intelligent replacement.
closed 2024-03-11 by yorikvanhavre