FreeCAD/FreeCAD #12826: [Problem] Context aware right click menu

maxwxyz, 2024-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the FreeCAD Day 2024 Complaint Session:

- **More things should be in the right click context menu**
I would like to see like right click get used to give more information and options in FreeCAD: more things in the right click context menu, as opposed to having to run all over the place with other buttons.
There should be relevant commants depending where the click happens and what is currently selected (tree, objects, geometry in the 3d view,...)

- **There are commands in the tree view that are not accessible within the 3d view**
Why is there the edit command object only in the tree view and not also in the 3D view?
For example, if you go in Part, draw a cube. If you right click in the tree view you have “edit cube”. But if you go in the 3D view on the cube, right click , there is no “edit cube.”

- **In tree view available toolbar functions are not available in the right click context menus**
In the right click context menu there's an option to create a part, but not an option to create a group folder.
When you right click on the tree view, you can create some objects but not others. And you can create all of them from the toolbar.

There should be a unified system where workbenches dock onto to change the context menu depending on the current workbench, selection and place where you right click (3d view, tree view, task panel, windows,...)

Currently the context menu is mostly static, displaying features which cannot be used and even are greyed out if the right click happens in the 3D view, on tree features or in other workbenches (e.g. spreadsheet).
The standard item in the 3D view looks like this:
[image]
If you change the workbench, mostly the right click menu stays the same.

The right click menu is currently overloaded with unusable commands for the selection.
If you right click on a spreadsheet object in the tree view, the following context menu pops up (was cut off at the bottom of the screen). All framed commands are useless for spreadsheets but are still displayed. Some spreadsheet related functions like *New spreadsheet*, *Import Spreadsheet*, *Export Spreadsheet* are missing.
[image]

Something similar was implemented in the Sketcher WB: https://github.com/FreeCAD/FreeCAD/pull/11884 There, the context menu is context aware, depending on the selection and preselection of the sketch.

Related issues regarding the context menu:
- https://github.com/FreeCAD/FreeCAD/issues/6043
- https://github.com/FreeCAD/FreeCAD/issues/11355
- https://github.com/FreeCAD/FreeCAD/issues/11429
- https://github.com/FreeCAD/FreeCAD/issues/7646
- https://github.com/FreeCAD/FreeCAD/issues/5877
- https://github.com/FreeCAD/FreeCAD/issues/12801
- https://github.com/FreeCAD/FreeCAD/issues/11413
- https://github.com/FreeCAD/FreeCAD/issues/9505

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Core

### Anything else?

In Blender you can type in most context menus to search for commands inside of them, which is related to: https://github.com/FreeCAD/FreeCAD/issues/8290

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

greggjaskiewicz, 2024-03-30:
I'll be looking at it, as soon as I can sort out my macOS build https://forum.freecad.org/viewtopic.php?p=749556#p749556

kaiwas, 2025-10-01:
I'd like to add that the menu structure isn't perfect in itself, but it also changes depending on the object/workbench type and the activation location (tree view/3D window).
The same menu items jump and dance.
See the image.

[image]

I described this problem in this issue, which is essentially a duplicate. https://github.com/FreeCAD/FreeCAD/issues/24299
I think we need to analyze possible solutions and create a more streamlined menu that, if it changes, will strive for consistency despite all the differences between workbenches.
Perhaps something like this (conceptually)

[image]