FreeCAD/FreeCAD #12827: [Problem] Addon manager UI: sort by field too narrow

maxwxyz, 2024-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The drop down menu to sort the result list in the addon manager is too narrow to read the contents. It's even worse with the translated text. 

No matter how wide the display is, its not readable unless it is selected. So it is only unreadable when the menu is pulled down.
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Addon Manager

### Anything else?

Not sure if this is related to @obelisk79 OpenDark.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-09:
@chennes FYI

benj5378, 2024-10-04:
@maxwxyz Good first issue

maxwxyz, 2024-10-23:
@MisterMakerNL with dark/light it does not show `...` but is cut off. 
[image]

MisterMakerNL, 2024-10-23:
[image]
classic:
[image]

It does not :P
I have seen more issues with QT not being able to define the correct text height/size especially with high dpi settings, this seems to cause all the "doesn't fit in the box" issues. For this reason I also removed all the text-height based dimension in the stylesheets.
I wonder if they fixed it in QT 6.
Or that it is a windows 11 issues, or it is linked to your dpi settings.
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 +2 (Git)
Build type: Release
Branch: pr/17249
Hash: 89fe80602ff80c514f70ecb203bf89f0fdc5c0fa
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: Darker_minimum.qss/FreeCAD Dark/
Installed mods: 
  * Assembly3 0.12.2
  * BIM021 2021.12.0
  * Dracula 0.0.5
  * fasteners 0.5.27
  * FeedsAndSpeeds 0.5.0
  * freecad_metal_workbench 0.0.1
  * Iconpack 2023.10.25
  * IconThemes (Disabled)
  * lattice2 1.0.0
  * OpenTheme 2024.8.17
  * OpenTheme.backup1720258469.8390777 2024.7.3 (Disabled)
  * OpenTheme.backup1722450713.1457298 2024.7.3 (Disabled)
  * Render 2024.8.4
  * scanObjects
  * sheetmetal 0.4.21

```