FreeCAD/FreeCAD #12829: [Problem] Core: Freeze function issues

maxwxyz, 2024-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@FlachyJoe implemented a great feature to *freeze* the current state and exclude from recompute: https://github.com/FreeCAD/FreeCAD/pull/12580

As @FEA-eng mentioned in the PR https://github.com/FreeCAD/FreeCAD/pull/12580#issuecomment-1978531015 it is currently not available in the PartDesign WB.
I've played a bit with this feature and have some troubles which should be easy to fix:

1. The freeze icon is not optimal. It covers the entire icon of the feature, body or object. It is not clear if you want to unfreeze something if this is a body or part container, a spreadsheet or PartDesign features. Maybe it can also be combined with the eye icon as the appearance of the object is *frozen*
2. Could it be placed in the context menu section at the bottom, where "Mark to recompute" and "Recompute" is located, as I understand it's related to that?
3. Could it be renamed? I did not figure out what this is doing and had to look through the PRs. 
Proposals: Exclude from recompute, Lock current state, 
5. Could it have a checkbox like "Active body" (at the top of the context menu), like: [ x ] Locked to see the current state in the context menu

Maybe this is all just my personal preference, @obelisk79 maybe ask DWG about this.

[image]
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-03-09:
@maxwxyz thank you very much for these tests :+1:

leoheck, 2024-10-26:
This is still an issue @maxwxyz @FlachyJoe. Can we reopen this issue?

After freezing an object... there is no way to unfreeze it or even freeze others.
The context menu item disapears.

Also, can we have the same item on the Edit menu since the feature does not show up in the context menu? 
[image]

Also, updating info here would be required if this feature is added to the Edit menu.
https://wiki.freecad.org/Std_ToggleFreeze

```
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git) AppImage
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
```

leoheck, 2024-10-26:
Here is the evidence... 
[Screencast from 2024-10-26 11-58-34.webm](https://github.com/user-attachments/assets/a6e46967-eaaa-4a40-b5bf-86e6007924a3)

leoheck, 2024-10-26:
After restarting FreeCad I cannot find the Freeze function anymore. Is there a specific way to enable this function somehow?

FEA-eng, 2024-10-26:
> After restarting FreeCad I cannot find the Freeze function anymore. Is there a specific way to enable this function somehow?

Aren't you in Part Design by any chance? There this feature is not accessible.

maxwxyz, 2024-10-26:
this issue was never closed...

leoheck, 2024-10-26:
> > After restarting FreeCad I cannot find the Freeze function anymore. Is there a specific way to enable this function somehow?
> 
> Aren't you in Part Design by any chance? There this feature is not accessible.

hm, I was using part design WB. Why is it not accessible with the PD selected? The freeze doesn't work with Bodies?

FEA-eng, 2024-10-27:
> hm, I was using part design WB. Why is it not accessible with the PD selected? The freeze doesn't work with Bodies?

It works on Bodies but there's a known issue with this option not being shown in PD. So you can apply it to Bodies from another workbench like Part.

PR #12856 (2024-12-06): [Core] Fix for frozen status
Fix #12829 

Ping @maxwxyz 

- [x] Frozen state does not persist on save and opening a file. If I freeze an object, alter it and save it, the alteration is not displayed. When reopening, the modified object is visible and the object is not frozen anymore (recompute at opening). The last modification before closing is now irreversible and lost, although the object was frozen and shouln't be altered.
**a 'freezed' object now prevent the document to be saved.**
- [x] If a PartDesign body is frozen, all nested features can be changed and alter the frozen body.
**linked [... cut]
closed 2024-12-06 by chennes