FreeCAD/FreeCAD #12831: [Problem] Addon manager: window resizing issues

maxwxyz, 2024-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Resizing the Developer Tools window (as it is too large for my screen) makes it disappear.

https://github.com/FreeCAD/FreeCAD/assets/6246609/9c6cb1e1-13df-4c6c-96b2-c62c1e8bdf5b

2. Resizing the Python dependencies window 

https://github.com/FreeCAD/FreeCAD/assets/6246609/e4685d20-5fcd-4e44-9d1e-1d7f8afae266

3. Resizing the addon manager vertically is slow and laggy

https://github.com/FreeCAD/FreeCAD/assets/6246609/f937ea3b-f64d-488b-bb0e-a00f3263d1bc

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-09:
@chennes FYI

chennes, 2024-03-11:
For future reference, things like this should be three issues -- these aren't related in the code, and will be fixed in separate PRs.

chennes, 2024-03-11:
> Resizing the Developer Tools window (as it is too large for my screen) makes it disappear.

It actually shouldn't be allowing you to resize vertically. So the actual bug is twofold: first, that resizing is being allowed at all (which I cannot reproduce), and second that the window is too large and should be redesigned