FreeCAD/FreeCAD #12832: [Problem] Addon manager: Readme links reveal blank page

maxwxyz, 2024-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you click on the links in a readme a blank page appears. Either the page should be displayed or opened in the default web browser.
Also, items like the translation status (icons on GitHub) are displayed as lists with links. Users might misclick as they expect a translated version of the readme but they also get a blank page 
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-03-11:
I cannot reproduce -- when I click on a link in the README it correctly loads in my system browser.

APEbbers, 2024-03-11:
Hi,

I can reproduce the problem however not every link is a problem. For example the link "user instructions" in Assembly 4 result in a blank page. However the link "quick assembly from scratch" opens in my browser.

chennes, 2024-03-11:
Is "user instructions" a link to an anchor?

APEbbers, 2024-03-11:
I'm not sure what you mean by anchor but in the picture below I've marked the hyperlinks:
[image]

chennes, 2024-03-11:
I get it -- it's a relative URL, for some reason that's not opening correctly. I'll try writing a processor that makes all of those absolute.

maxwxyz, 2024-03-11:
@chennes relative URLs are the only one working on my side inside the Addon manager. It does not open external ones.

chennes, 2024-03-11:
@maxwxyz so I believe what you mean is that internal (anchor) links are working and simply scroll the page, but relative links to other pages are not working, they load a blank page in the addon manager. And presumably full-url links are working correctly and open in an external browser. Is that the case?

maxwxyz, 2024-03-11:
@chennes yes, internal anchors work. relative (GitHub internal pages) reveal blank pages. Some other external links also show a blank page. Try the AeroFoil macro readme page and click on the `Addon manager` link. This is a link to the wiki but a blank page appears.

PR #12893 (2024-03-13): Addon Manager: Fix relative external links
External links to relative files were not being followed because Qt had no way of knowing how to get to the file. Complicating matters, for links to markdown files we don't want the "raw" page (showing the markdown) to be the thing loaded in your browser, we want the rendered version. Fixes #12832.
closed 2024-03-13 by chennes