FreeCAD/FreeCAD #12836: [Problem] Draft Move of a linked object which belongs to a Std Part can result in wrong placement

JGBeu, 2024-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This will happen if the containing Std Part is in some way rotated from its original position.

Steps to reproduce:

1. Create a Std Part
2. Create a Part Box inside Part
3. Rotate Part 90° around z-axis
4. Create a Part Cylinder inside Part
5. Create a Std Link of Cylinder; this will create Link at the top level
6. Drag an drop Link onto Part, effectively adding Link to Part
7. Deselect everything
8. Switch to draft workbench
9. Change the draft working plane to XY (Top View)
10. Select Link
11. Select Draft Move, e.g. in relative mode
12. Select or enter origin e.g. by snapping to top center of Link
13. Select or enter destination with delta x <> 0 or delta y <> 0 e.g. by snapping to some different corner of Box

Expected behaviour is that the new postition of Link is at the selected destination.
Observed behaviour is that the new position of Link is at the selected destination rotated 90° aroud z-axis

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36331 (Git) AppImage
Build type: Release
Branch: main
Hash: eb5a1b333946ce4ac8a4ead835cba8bdcf602a87
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Render 2023.8.18
  * Help 1.0.3
  * POV-Ray-Rendering
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-10:
It works for me without a problem
[image]
Could you upload your file?
<details><summary>Windows 11 0.22dev</summary>
<p>

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

</p>
</details>

JGBeu, 2024-03-10:
Sure. I am on Linux, mybe that's different.
[Issue.zip](https://github.com/FreeCAD/FreeCAD/files/14549495/Issue.zip)

JGBeu, 2024-03-10:
No, just checked with the 22.0dev on Windows 11, same issue.
Difference is that in your setup Link Cylinde001 is outside Part. The problem arises if Part first contains the Link, and then Link is moved.

maxwxyz, 2024-03-10:
No, works the same with your file on my end: 
[image]

I've moved the link outside and then moved it. Do you have snapping on?

maxwxyz, 2024-03-10:
Or do you mean that is moves after you've moved it inside the Part Container and then dragged it outside of the Part?
That's an expected behavior as the contents of the Part Container are positioned relative to the part origin / axis sysem. If you move/rotate this, all features inside move with it. If you put it outside again they have their own (prior) position.
https://wiki.freecad.org/Std_Part/de

JGBeu, 2024-03-10:
I can reproduce it with snapping on as well as off. Moving the link outside helps in the first place. But if you move it back into the part it jumps to a different place due to translation of the part itself.
Bottomline is that Draft Move does not take into account the orientation of the container. Which renders links impractical to use when designing with assemblies.

maxwxyz, 2024-03-10:
It has nothing to do with links, that is just the way a part container works, otherwise you could not group the items and move them together.
Try it if you copy the cylinder instead of making a link, same behavior. And this is not just the Draft Move but everything which changes the orientation.

JGBeu, 2024-03-10:
The link behaves different from e.g. a duplicate of the part. If you duplicate Cylinder the duplicate stays in the container and Draft Move behaves different on the new cylinder inside the container than it would on the link.
That's why for now I always work with duplicates inside assemblies, unless I can be sure to not have to change positions of the objects in the container afterwards.

JGBeu, 2024-03-10:
Sorry, did not notice you closed the issue. Thank you for your time.

maxwxyz, 2024-03-10:
Could you make a screen recording or somehow demonstrate the issue and difference from link to a copy?

JGBeu, 2024-03-10:
I'll try a recording.

JGBeu, 2024-03-10:
There you go...
[image]

maxwxyz, 2024-03-10:
Ok that's weird, I cannot reproduce this but it should work as its all inside of the PartBody.

maxwxyz, 2024-03-10:
@JGBeu what happens if you use the [Std_TransformManip](https://wiki.freecad.org/Std_TransformManip)? is is also moved to a different location after applying?

JGBeu, 2024-03-10:
TransformManip works as expected. The issue is only with Draft Move. But transform manipulator lacks the ability of aligning objects at snap points, which I consider often essential.

Roy-043, 2024-03-11:
Confirmed. Draft_Move does not take the nesting into account.

PR #18795 (2025-01-20): Draft: make move, rotate and scale commands link-aware
Fixes #12836.
Fixes #15681.

These changes are presented as a single ('big') PR because the mentioned commands have several similarities. Working on them seperately would have made less sense.

The commands have been made 'link-aware' meaning they can now handle Links and objects in linked containers.

This required several changes in the following main files. For each command all options are handled by a single function now (the `move`, `rotate` and `scale` functions). This was the only reasonable solution to correctly handle nested place [... cut]
commit d587e6a9 (2025-01-20): Draft: make move, rotate and scale commands link-aware (#18795)

* Draft: make move, rotate and scale commands link-aware

Fixes #12836.
Fixes #15681.

These changes are presented as a single ('big') PR because the mentioned commands have several similarities. Working on them seperately would have made less sense.

The commands have been made 'link-aware' meaning they can now handle Links and objects in linked containers.

This required several changes in the following main files. For each command all options are handled by a single function now (the `move`, `rotate` and `scale` functions). Th [... cut]
closed 2025-01-20 by yorikvanhavre