FreeCAD/FreeCAD #12837: [Problem] Edit Custom Property and sorting

kaiwas, 2024-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sometimes I have to write my own properties. If there are a lot of them, unfortunately there are mistakes in their writing. But at the moment there is no way to quickly correct the inaccuracy. Because "edit property" is missing.

You have to delete the previously created property and create it again to correct the error.

If there was an “edit property” tab that would open a window similar to the one that opens when you click “add property,” but with the data for this property already filled in previously, that would be very convenient. (picture number 1).
It would be possible to change the property name and its description (the group can be changed now)

[image]

Another problem is the order of properties in the panel. Now there is no way to sort them in the desired order (the order is strictly alphabetical). We have to come up with various workarounds so that the necessary properties are in the right order and are always at the top of the panel (picture number two)

[image]

### Full version info

```shell
Any version
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-10:
related: #12120

kaiwas, 2024-03-10:
> related: #12120

Sorry if I created a duplicate. I searched for the problem, but it seems I used the wrong terms in the search.
Thank you

maxwxyz, 2024-03-10:
I think it is not a duplicate but still related. Currently you cannot sort or edit the name of the properties as they are currently implemented. @pieterhijma ping to consider this as well.

pieterhijma, 2024-03-11:
Yes, very good point. It's not a duplicate. I'll look into that.

pieterhijma, 2024-09-06:
Changing the order appears to be a much requested feature, I know of two others who would like this.

kaiwas, 2024-09-06:
> Changing the order appears to be a much requested feature, I know of two others who would like this.

I had another idea that would significantly improve access to properties.
But this topic was closed and moved to discussions that no one will ever see )
https://github.com/FreeCAD/FreeCAD/discussions/14026

furgo16, 2024-10-07:
Also related, but for VarSets: https://github.com/FreeCAD/FreeCAD/issues/16222