FreeCAD/FreeCAD #12839: [Problem] Double click issues in the tree view

maxwxyz, 2024-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the FreeCAD Day 2024 Complaint Session:
**Double click in tree view is inconsistent is overloaded, and doesn’t communicate what it will do when you double click**
Double clicking in the tree is inconsistent. There's at least five different actions that will happen if you double click everything from changing a label to activating transform, to opening task panel or changing workbenches.
Double click is overloaded and it doesn't communicate what it's going to do when you double click and what it has done with the double click.

There is also the [Std_UserEditMode](https://wiki.freecad.org/Std_UserEditMode) but it seems not to work reliable in the latest builds, is not supported by all objects and the function is unclear.
Also, the current mode is not reflected anywhere, it seems nothing is happening when changing the mode.
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Core

### Anything else?

Related:

- https://github.com/FreeCAD/FreeCAD/issues/12124
- https://github.com/FreeCAD/FreeCAD/issues/11429

https://forum.freecad.org/viewtopic.php?t=69579
https://forum.freecad.org/viewtopic.php?p=609025

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xerrio, 2026-09-04:
Today I did a square pad. Then I edited the sketch. Then I double-clicked on the padding... but it openned the "transform" window instead of "Pad parameters". I was forced to right-click on the "pad" then to left click on "edit pad" in the context menu.

Annoying. I spent a bit of time understanding why sometimes a double-click on a pad in the tree opens the pad parameters window and sometimes something else.

FEA-eng, 2026-10-16:
> 14\. Today I did a square pad. Then I edited the sketch. Then I double-clicked on the padding... but it openned the "transform" window instead of "Pad parameters".

Make sure that you have Edit Mode set to Default.