FreeCAD/FreeCAD #12840: [Problem] Build error in TopoShapePyImp.cpp

waarmond, 2024-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The build of current git with gcc 13.2.1 (boost 1.84.0) fails:

FAILED: src/Mod/Part/App/CMakeFiles/Part.dir/TopoShapePyImp.cpp.o 
/usr/bin/ccache /usr/lib/ccache/bin/c++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DFCUseFreeType -DFMT_SHARED -DHAVE_CONFIG_H -DPYCXX_6_2_COMPATIBILITY -DPart_EXPORTS -DQT_CORE_LIB -DQT_NO_DEBUG -DQT_NO_KEYWORDS -DQT_XML_LIB -D_OCC64 -I./FreeCAD/build/src/Mod/Part/App/Part_autogen/include -I./FreeCAD/build -I./FreeCAD/build/src -I./FreeCAD/src -I./FreeCAD/build/src/Mod/Part/App -I./FreeCAD/src/Mod/Part/App -I/usr/include/python3.10 -I/usr/include/freetype2 -isystem /usr/include/opencascade -isystem /usr/include/qt -isystem /usr/include/qt/QtCore -isystem /usr/lib/qt/mkspecs/linux-g++ -isystem /usr/include/qt/QtXml -Wall -Wextra -Wno-write-strings -march=native -mtune=native -O3         -pipe -fno-plt -fexceptions -Wformat         -w -Wno-deprecated-declarations -fno-permissive -Wp,-D_GLIBCXX_ASSERTIONS -fpermissive -Wzero-as-null-pointer-constant -fPIC -w -std=gnu++17 -fPIC -fPIC -MD -MT src/Mod/Part/App/CMakeFiles/Part.dir/TopoShapePyImp.cpp.o -MF src/Mod/Part/App/CMakeFiles/Part.dir/TopoShapePyImp.cpp.o.d -o src/Mod/Part/App/CMakeFiles/Part.dir/TopoShapePyImp.cpp.o -c ./FreeCAD/src/Mod/Part/App/TopoShapePyImp.cpp
In file included from ./FreeCAD/src/Mod/Part/App/TopoShapePyImp.cpp:95:
./FreeCAD/src/Mod/Part/App/ShapeMapHasher.h: In member function 'size_t Part::ShapeMapHasher::operator()(const TopoDS_Shape&) const':
./FreeCAD/src/Mod/Part/App/ShapeMapHasher.h:43:40: error: could not convert '<brace-enclosed initializer list>()' from '<brace-enclosed initializer list>' to 'std::__hash_enum<TopoDS_Shape, false>'
   43 |         return std::hash<TopoDS_Shape>{}(theShape);
      |                                        ^
      |                                        |
      |                                        <brace-enclosed initializer list>
./FreeCAD/src/Mod/Part/App/ShapeMapHasher.h:43:40: error: use of deleted function 'std::hash<TopoDS_Shape>::~hash()'
In file included from /usr/include/c++/13.2.1/string_view:42,
                 from /usr/include/c++/13.2.1/bits/basic_string.h:47,
                 from /usr/include/c++/13.2.1/string:54,
                 from /usr/include/c++/13.2.1/bits/locale_classes.h:40,
                 from /usr/include/c++/13.2.1/bits/ios_base.h:41,
                 from /usr/include/c++/13.2.1/ios:44,
                 from /usr/include/c++/13.2.1/istream:40,
                 from /usr/include/c++/13.2.1/sstream:40,
                 from ./FreeCAD/src/Mod/Part/App/TopoShapePyImp.cpp:25:
/usr/include/c++/13.2.1/bits/functional_hash.h:102:12: note: 'std::hash<TopoDS_Shape>::~hash()' is implicitly deleted because the default definition would be ill-formed:
  102 |     struct hash : __hash_enum<_Tp>
      |            ^~~~
/usr/include/c++/13.2.1/bits/functional_hash.h:102:12: error: 'std::__hash_enum<_Tp, <anonymous> >::~__hash_enum() [with _Tp = TopoDS_Shape; bool <anonymous> = false]' is private within this context
/usr/include/c++/13.2.1/bits/functional_hash.h:84:7: note: declared private here
   84 |       ~__hash_enum();
      |       ^
[1690/4379] Building CXX object src/Mod/Part/App/CMakeFiles/Part.dir/ChFi2d/ChFi2d_FilletAlgoPyImp.cpp.o

### Full version info

```shell
git master rev 5b5be480a3a14963cbdefe2e7a3923f9136af6a9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-03-10:
What is your OS? What version of OCCT are you trying to compile against? Do you compile with C++17 or C++20 on?

What is your _cmake_ configuration?

waarmond, 2024-03-10:
- ArchLinux
- OCCT 7.7.2
- C++17

```
  cmake -Wno-dev -G Ninja -B build -S . \
    -D BUILD_ENABLE_CXX_STD=C++17 \
    -D BUILD_QT5=ON \
    -D BUILD_FEM=OFF \
    -D BUILD_SMESH=OFF \
    -D FREECAD_USE_EXTERNAL_SMESH=OFF \
    -D BUILD_SHIP=ON \
    -D BUILD_ASSEMBLY=OFF \
    -D BUILD_PLOT=ON \
    -D CMAKE_BUILD_TYPE=None \
    -D CMAKE_C_FLAGS="${CFLAGS} -fPIC -w" \
    -D CMAKE_CXX_FLAGS="${CXXFLAGS} -fPIC -w" \
    -D FREECAD_USE_EXTERNAL_PIVY=ON \
    -D FREECAD_USE_OCC_VARIANT="Official Version" \
    -D FREECAD_USE_QT_FILEDIALOG=ON \
    -D INSTALL_TO_SITEPACKAGES=ON \
    -D CMAKE_INSTALL_PREFIX="${_destdir}/lib/freecad" \
    -D CMAKE_INSTALL_BINDIR=bin \
    -D CMAKE_INSTALL_LIBDIR='../../lib' \
    -D CMAKE_INSTALL_DATADIR='../../share/freecad' \
    -D CMAKE_INSTALL_DATAROOTDIR='../../share' \
    -D CMAKE_INSTALL_DOCDIR='../../share/doc/freecad'
```

xtemp09, 2024-03-10:
Could you kindly provide also `${CFLAGS}`, `${CXXFLAGS}` and your OS version?

I've never encountered `CMAKE_BUILD_TYPE=None`. 😢  Does _Release_, _RelWithDebInfo_ or _Debug_ compile?

xtemp09, 2024-03-10:
I just don't think that the configuration supports this option. If you look at _cmake_ files, you will see that _cmake_ adds some options if the build is _Release_, _Debug_, _RelWithDebInfo_, _conda_ and others. It seems to me that if you use 
`CMAKE_BUILD_TYPE=None` you have to manually add some parameters that are automatically added otherwise. [^1] [^2]

For example, `-DNDEBUG` option is absent in the command-line, meaning that it's a debug build, but `-march=native -mtune=native -O3` contradicts it.

[^1]: https://github.com/FreeCAD/FreeCAD/blob/main/CMakePresets.json
[^2]: https://github.com/FreeCAD/FreeCAD/tree/main/cMake

bgbsww, 2024-08-11:
@maxwxyz  not Topic: Toponaming - it's a simple build issue.  And it looks like a completely unsupported one that should have this closed as a won't fix.

closed 2024-08-11 by maxwxyz