FreeCAD/FreeCAD #12841: [Problem] Quickly convert external geometry to base sketch geometry.

kaiwas, 2024-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I understand that working with external geometry until the toponaming problem is fixed is not very relevant, but I would still like to voice it.

1. There is currently no quick way to convert external geometry to base sketch geometry. To solve this problem, you have to outline the external geometry by clicking on all the vertices of the external geometry. Moreover, this procedure is very complicated because it requires excessive precision to hit the vertices.

Maybe it would be possible to make some kind of tool, when you click on it, the line of the external geometry would turn into a base line (probably it would simply automatically create a stroke)? (I tried different options but didn't find a solution)

[image]

2. It seems that external geometry cannot always be created from faces. This can be seen at the initial stage in the attached video. Although, in essence, the selected plane has restrictions, which we can ultimately obtain by clicking on each edge of this face.

https://github.com/FreeCAD/FreeCAD/assets/32273833/3bdddb17-3ea6-4b90-a670-b2b67feb6b1f

### Full version info

```shell
Any Version
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-10:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/10043

maxwxyz, 2024-03-10:
@kaiwas you could add your description and screenshots as a comment in https://github.com/FreeCAD/FreeCAD/issues/10043

maxwxyz, 2024-03-10:
and point 2 is: https://github.com/FreeCAD/FreeCAD/issues/12670

closed 2024-03-10 by maxwxyz

kaiwas, 2024-03-10:
> Duplicate of #10043

There are so many requests and given the difficulties of translation, I could not find this problem earlier. The search did not show the required option. (sorry for the duplicate).

I have moved my text. This thread can be deleted if necessary)

maxwxyz, 2024-03-10:
I've closed it already, all good.