FreeCAD/FreeCAD #12842: [Problem] Recent file list issues

maxwxyz, 2024-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the FreeCAD Day 2024 Complaint Session:
**There is no obvious way to clear recent file list**
There's no obvious way to clear the recent part. You should be able to remove the entire history or remove individual entries.

**Recent file list is far to aggressive, it contains files reference by project**
The current implementation of recently opened files is far too aggressive. Open up a document with a bunch of linked documents it will swamp your recent file list with everything.
So the recent file history contains documents that you didn't explicitly choose to open, but for reference by other files.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Core

### Anything else?

Related: #12565

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-03-10:
> The current implementation of recently opened files is far too aggressive. Open up a document with a bunch of linked documents it will swamp your recent file list with everything.
> So the recent file [...]

@PaddleStroke is this getting fixed with #12406?

PaddleStroke, 2024-03-10:
I don't think so because OP does not mention assembly. So I guess this
behavior also happens with partial loading (though I can't test right now)

jorgebr, 2024-08-30:
For simple users just searching for a way how to do this, even if it's not part of the user interface:
Go to Tools > Edit Parameters ...
Then BaseApp > Preferences > RecentFiles

You can edit or delete any entries there. You must restart FreeCAD to see the changes.

chibiconsulting, 2025-02-22:
> 
> From the FreeCAD Day 2024 Complaint Session: **There is no obvious way to clear recent file list** There's no obvious way to clear the recent part. You should be able to remove the entire history or remove individual entries.
> 
> **Recent file list is far to aggressive, it contains files reference by project** The current implementation of recently opened files is far too aggressive. Open up a document with a bunch of linked documents it will swamp your recent file list with everything. So the recent file history contains documents that you didn't explicitly choose to open, but for reference by other files.
> 
> It is also beneficial to pin certain Documents to the top of the recent file list e.g. favorites or templates.

A suggestion that could help to reduce the impact of this problem is to add a user selected filter for which file types are included in the recent file list. This could be a list of the sypported file types with a check box next to each type.

For me, I would only like to see .FCStd files so I would uncheck the other supported file types from the filter list. Other people might only want to see recent .STL files or recent .DWG files so they could select those that they would like to see.

For example:

Recent File List Filter

_ All Files
_ FreeCAD (*.FCStd)
_ 3D Studio mesh (*.3ds)
...
_ Web page (*.html, *.xhtml)

Also, a vote for the option to pin a project to the RFL, as that could help some times as well.

xtemp09, 2025-07-24:
> 
> It is also beneficial to pin certain Documents to the top of the recent file list e.g. favorites or templates.
> 

I have never seen software utilizing this concept. @maxwxyz, can you provide an example or a screenshot?

xtemp09, 2025-07-25:
> But this can also be done now with custom directories at the start page.

If this can be done with _Start_ page, what is exactly you are talking about?

When I look at examples, I usually choose _Qt Creator_. This program has recent files list similar to _FreeCAD_. In fact, most programs have a similar interface.

What you're referring to is likely jump list. Implementing this requires significant changes to the interface — it's not just a simple menu item update.

xtemp09, 2025-08-02:
@maxwxyz, could you adjust the issue according to the recent events? It looks like you're referring to jump list, but this is supposed to be a part of _Start_ page widget.

luzpaz, 2025-11-05:
@xtemp09 would you recommend a title ITC ?

luzpaz, 2025-11-16:
'In that case'

ipatch, 2026-08-15:
@maxwxyz with the current `clear recent files` recent file list the ellipse got removed from the text. it was my understanding from the original PR i submitted that ellipse aid the user in letting them know a separate / new window will open when selecting the menu entry. there was a whole discussion about when i submitted my PR.

- https://github.com/FreeCAD/FreeCAD/pull/27274

ie.

[image]

and now the menu entry looks like the following.

[image]

and ended coming across all of this when reading through,

https://github.com/FreeCAD/FreeCAD/issues/31927#event-29464404782

maxwxyz, 2026-08-16:
@ipatch yes this is correct, see https://github.com/FreeCAD/FreeCAD-Enhancement-Proposals/blob/master/FEPs/FEP-0007-consistent-language/README.md

> In general, limit the use of ellipsis. Use an ellipsis only when the action requires further user input, where the ellipsis stands for the missing part and the command or context would benefit from it:
> 
>     Example: Export As… → asks for a file type/name.
>     Example: Save As… → asks for a file name.
> 
> Do not use ellipses merely because an action opens a dialog:
> 
>     Bad: Unit Converter… (no further text to complete the phrase)
>