FreeCAD/FreeCAD #12843: [Problem] There is no 3D PDF export option

maxwxyz, 2024-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is no way to directly export into a 3D PDF and keep tree structure, materials and custom views.
https://helpx.adobe.com/acrobat/using/adding-3d-models-pdfs-acrobat.html

3D related file types to be used in 3D PDF are Universal 3D (.U3D) files (https://en.wikipedia.org/wiki/Universal_3D)

Related issue for import of these files:
- https://github.com/FreeCAD/FreeCAD/issues/6090

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Core

### Anything else?

https://helpx.adobe.com/acrobat/using/displaying-3d-models-pdfs.html
https://helpx.adobe.com/acrobat/using/interacting-3d-models.html#interacting_with_3d_models
https://helpx.adobe.com/acrobat/using/measuring-3d-objects-pdfs.html#measuring_3d_objects_in_pdfs

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-03-20:
Indeed, 3D PDF export requires a 3rd party software and even after Meshlab bug was fixed, it's still unnecessarily tedious: https://forum.freecad.org/viewtopic.php?t=81074&start=10#p726018

An option to export 3D PDF files directly from FreeCAD would be nice. I've seen several users asking for this.

chiragsingh1711, 2025-03-23:
Hello @maxwxyz  ,

I'm Chirag Singh, and I'm interested in contributing to this project. I have extensive experience with 3D software like Blender, Fusion360, and FreeCAD, specializing in automation scripting. While I haven't worked directly with 3D PDFs, I'm eager to apply my skills and learn new technologies for this project.

Could we discuss how my experience aligns with your needs and how I can best contribute?

Best regards,
Chirag Singh

maxwxyz, 2025-03-23:
Sure. I'm not an expert but maybe @pieterhijma knows about OCC export to 3D PDF or maybe @WandererFan for the TechDraw PDF exports.

chiragsingh1711, 2025-03-25:
Hi @pieterhijma & @WandererFan ,

I'd like to discuss the implementation of a 3D PDF export feature for FreeCAD for GSoC'25, a feature that many users have been requesting. This feature is crucial for several reasons:

- Users can have enhanced collaboration with non-CAD users.
- Improved documentation and presentation of 3D designs to clients.

and more ......

In my research, I've found that other CAD software, such as AutoCAD, already offer this functionality. Specifically, **ProtoTech Solutions** has developed a 3D PDF Exporter plugin for AutoCAD with the following architecture:

- Traverses native data structures
- Converts 3D models to PRC format
- Generates PDF with embedded 3D content
- Includes a template system for customization

**Key components include:**

- Native API integration
- Tessellation engine
- PRC encoder
- PDF generator
- User interface for export options

**For FreeCAD, I propose a similar approach:**

- Use FreeCAD's Python API to access the document structure
- Convert Coin3D scene graph to PRC format:
- Traverse the Coin3D scene graph
- Extract geometry, materials, and transformation data
- Convert to PRC format
- Generate PDF with embedded 3D content using a library like libHaru

I believe this approach aligns with FreeCAD's architecture while providing the necessary functionality. However, I'd greatly appreciate the community's feedback:

- **Is this approach feasible and in line with FreeCAD's development philosophy?**
- **Are there any specific challenges or considerations I should be aware of?**
- **Are there any existing efforts or components within FreeCAD that could be leveraged for this project?**

Your insights and opinions are invaluable in ensuring we're moving in the right direction. Thank you for your time and expertise.

pieterhijma, 2025-03-25:
Hi @chiragsingh1711, that would be a great project!

It is difficult for me to judge whether this is feasible. Firstly, I don't think the scene graph is necessary, it is probably enough to know the hierarchy of the document objects. 

Secondly, I think converting it to the PRC or U3D format is going to be the challenging part and the best way would probably be to use support from OpenCASCADE. I tried to look it up, but it seems that OpenCASCADE does not have support for these file formats yet. Converting it yourself is probably quite an undertaking.

Existing efforts is perhaps [this forum thread](https://forum.freecad.org/viewtopic.php?t=95294) on OCCT data exchange. Perhaps OCCT has a module that supports PRC.

My advise would be to first study how exactly the conversion should happen to PRC, U3D, or any other format that PDF supports for 3D objects. Does this help?

chiragsingh1711, 2025-03-26:
@pieterhijma 
Thank you for your valuable feedback on my proposal. I've conducted thorough research on the conversion process as you suggested, and I'd like to share my findings.

You were absolutely right that focusing on traversing document object hierarchy rather than the scene graph is the better approach. 

Regarding the PRC/U3D conversion challenge, I've identified two potential pathways:

1. **Asymptote Integration**: The most promising approach appears to be leveraging Asymptote's PRC implementation. This software has a robust PRC writer that supports NURBS surfaces and complex assemblies. Several forum posts indicate FreeCAD users are already using Asymptote with LaTeX for 3D PDF creation, though in a multi-step process.

link 1 - https://forum.freecad.org/viewtopic.php?t=48915
link 2 - https://forum.freecad.org/viewtopic.php?style=3&p=428909
link 3 - https://forum.freecad.org/viewtopic.php?style=5&p=431875

Also someone in the community posted a video tutorial  - https://www.youtube.com/watch?v=Q_ufaCN2hb4

2. **OpenCASCADE Pathway**: While OpenCASCADE doesn't have native PRC support, I found a detailed blog post describing a workflow for translating OpenCASCADE geometry to 3D PDF via PRC. This suggests it's technically feasible to build this bridge.

Link - https://www.cnblogs.com/opencascade/p/5317745.html (Its in Chinese, had to translate and understand)

Would you agree with this direction? I'd appreciate your thoughts on whether focusing on Asymptote integration first would be the most practical approach, or I should explore OpenCASCADE Pathway. Though I'll try both ways and do a proper comparison before working on the proposal.

maxwxyz, 2025-03-27:
thanks, would be appreciated very much if you would push it to FreeCAD!

chiragsingh1711, 2025-03-27:
Thank you @marekr for sharing your KiCad implementation! This is a valuable reference. Though I'd like to pursue this for my GSoC and was proposing to implement support for PRC (by converting the part to mesh to .asy and finally to prc), plus some additional pdf customizable options discussed in my proposal on FreeCAD Forum.

I have a question, why did you choose to implement U3D rather than PRC? Is PRC feasible to implement or U3D is a better option ?

marekr, 2025-03-27:
> Thank you [@marekr](https://github.com/marekr) for sharing your KiCad implementation! This is a valuable reference. Though I'd like to pursue this for my GSoC and was proposing to implement support [...]

I don't get paid to do this work, so I'm not buying the ISO 14739-1:2014 standard for 221 swiss francs lol

U3D is a ECMA standard so it's free.