FreeCAD/FreeCAD #12845: [Problem] Relation between document and file name not fixed

maxwxyz, 2024-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the FreeCAD Day 2024 Complaint Session:
****

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-10-04:
I don't really get the issue here? Isn't it a feature, that label and filename doesn't have to be same? Do anyone want label and filename to be the same?

maxwxyz, 2024-10-05:
Not OP but I guess the idea was that the document should be the same as file name.

benj5378, 2024-10-05:
> Not OP but I guess the idea was that the document should be the same as file name.

I mean, I consider it as an intended feature, that document name and file name can be seperate - else the document name property wouldn't have been made in the first place.

Can someone weigh in here, if this is considered an issue or should be closed? @obelisk79 @qewer33