FreeCAD/FreeCAD #12846: [Problem] No default selection set for file export

maxwxyz, 2024-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the FreeCAD Day 2024 Complaint Session:
**There is no default selection set for export even if you only have one item**
In the export file I select exactly one thing and I click export, it works. If you have no selected anything it won't work.
There there is no default selection set. For an export, even if you only have only one single item in the document.

The export should ask for a selection if no selection is present when clicking on export. If only one item is present in the tree, the item should be selected if feasible for the desired export file format.
If you have multiple items, e.g. a TechDraw page and a body and you select nothing and try to export a .STEP the body should be selected. If you tried to export a technical drawing PDF the TechDraw page should be selected (and so on).

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maritaria, 2024-12-05:
Relevant code section: https://github.com/FreeCAD/FreeCAD/blob/870e260bc926e50651507c905eaceb5576a2da37/src/Gui/CommandDoc.cpp#L413

maritaria, 2025-01-05:
Alright, after some digging through the codebase, the patch at the bottom implements parts of the issue.
Mainly that the export command does *something* when attempting to export an empty selection.
This patch replaces the empty selection with all objects of the active document.
Also, I changed the function which builds the file name to receive the list of document objects instead of doing its own call to get the current selection.

This patch does not cover functionality that the export type should influence what sub-set of all objects should be supported. But perhaps the requirement is already satisfied by the existing export mechanism (one can hope).

Importantly, I have not yet been able to test this code yet, as I've not yet set up my local machine to compile FC from scratch. I work slowly on this in my spare time, of which I have little. So it may be another month or two until I have compilation set up and have been able to give this patch an actual test. At which point I'll be able to send in a PR of code which I know actually works. 

I'm posting this as progress so anyone with spare time on their hand has the chance to pick up where I left off.

```diff
diff --git a/src/Gui/CommandDoc.cpp b/src/Gui/CommandDoc.cpp
index f6791360ad..a1e2451176 100644
--- a/src/Gui/CommandDoc.cpp
+++ b/src/Gui/CommandDoc.cpp
@@ -343,11 +343,10 @@ The format string is stored in two user preferences (not currently exposed in th
 * BaseApp/Preferences/General/ExportDefaultFilenameSingle
 * BaseApp/Preferences/General/ExportDefaultFilenameMultiple
 */
-QString createDefaultExportBasename()
+QString createDefaultExportBasename(const std::vector<App::DocumentObject*>& selection)
 {
     QString defaultFilename;
 
-    auto selection = Gui::Selection().getObjectsOfType(App::DocumentObject::getClassTypeId());
     QString exportFormatString;
     if (selection.size() == 1) {
         exportFormatString = QString::fromStdString (App::GetApplication().GetParameterGroupByPath("User parameter:BaseApp/Preferences/General")->
@@ -457,6 +456,10 @@ void StdCmdExport::activated(int iMsg)
     static Document* lastActiveDocument;
 
     auto selection = Gui::Selection().getObjectsOfType(App::DocumentObject::getClassTypeId());
+    if (selection.empty()) {
+      selection = App::GetApplication().getActiveDocument()->getObjectsOfType(App::DocumentObject::getClassTypeId());
+    }
+
     if (selection.empty()) {
         QMessageBox::warning(Gui::getMainWindow(),
             QCoreApplication::translate("StdCmdExport", "No selection"),
@@ -510,7 +513,7 @@ void StdCmdExport::activated(int iMsg)
         }
 
         if (lastExportUsedGeneratedFilename   || didActiveDocumentChange) {  /*<- static, true on first call*/
-            defaultFilename = defaultExportPath + QLatin1Char('/') + createDefaultExportBasename();
+            defaultFilename = defaultExportPath + QLatin1Char('/') + createDefaultExportBasename(selection);
 
             // Append the last extension used, if there is one.
             if (!lastExportFullPath.isEmpty()) {
```

To apply this patch on your machine, first paste the snippet into a new file somewhere in the repository and then run `git apply $path_to_diff_file`.