FreeCAD/FreeCAD #12847: [Problem] Snapping should be available globally

maxwxyz, 2024-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the FreeCAD Day 2024 Complaint Session:
**Snapping only works inside draft**
Draft snapping and modification tools should be globally available. 
Snapping only works inside the DraftWB and it should work everywhere.

Snapping tools should be available globally and be able to use in any workbench if possible. Examples would be to snap gemetry inside sketches, snap to positions while moving assemblies, snap when transforming using the Std_TransformManip, snapping when arragning TechDraw dimensions, when interactive tools are available: snap to define the extrude length in the 3d view,...

Related issues to snapping:
- https://github.com/FreeCAD/FreeCAD/issues/10072
- https://github.com/FreeCAD/FreeCAD/issues/6240
- https://github.com/FreeCAD/FreeCAD/issues/6026
- https://github.com/FreeCAD/FreeCAD/issues/5557
- https://github.com/FreeCAD/FreeCAD/issues/11636
- https://github.com/FreeCAD/FreeCAD/issues/10599
- https://github.com/FreeCAD/FreeCAD/issues/10600
- https://github.com/FreeCAD/FreeCAD/issues/6204
- https://github.com/FreeCAD/FreeCAD/issues/11173
- https://github.com/FreeCAD/FreeCAD/issues/11169

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-03-11:
Sketcher has snapping to object. There are some features missing like snapping to intersection.

Having said that it's true that snapping could be made more global. In sketcher it's currently implemented in the viewprovidersketch by overriding the mouse position. This could be moved in Part module where it would be able to share between draft/sketcher/assembly.

yorikvanhavre, 2024-03-11:
Agree! That would be cool, and probably much faster than the current Draft system.
One could start simply by rewriting the Draft snap system in C++ :) It's pretty much self contained in one file: https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Draft/draftguitools/gui_snapper.py
But this is maybe the opportunity to think to something better