FreeCAD/FreeCAD #12864: Default view buttons take away too much space

maxwxyz, 2024-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The default view buttons (iso, front, top, right, rear, bottom, left) have each an individual icon in the task bar and take away a lot of space, especially on smaller screens. 
[image]

I propose these could be grouped in a command group (like selection filter or draw style) as this is also done in other applications e.g. CATIA: 
[image]

These 7 buttons could be reduced to just one. It is still possible to quickly change to default views:

- using the shortcut keys: 1,2,3,...
- using the navicube
- using the right click menu
- using the menu
- using this new command group

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2024-03-11:
I am generally in favor of such a change. However a few thoughts and concerns I have are such:
1. There needs to be a really good icon design to convey the intent of the group
2. I foresee a lot of community pushback on this kind of change
3. A preference to switch between new vs old would be undesirable

An option would be to add the command group as the primary in the view toolbar and have the discrete buttons in a more verbose disabled-by-default std_views toolbar.

pierreporte, 2024-03-11:
Why not putting all the views in the “cube” menu of the navigation cube?

[image]

maxwxyz, 2024-03-11:
There is a PR for that but some people have the cube disabled.
https://github.com/FreeCAD/FreeCAD/pull/12354

Mannshoch, 2024-03-13:
@maxwxyz May disable the possibility to disable the cube and replace it with a setting to have only a smaller cube with may less function.

PR #12878 (2024-03-18): UI/UX: Create a toolbar command group for default views
fixes #12864 
This PR creates a grouped command button in the toolbar for all the default views (more in the issue). It is enabled by default.
All previously visible individual view buttons are available as a separate toolbar and can be easily enabled.

Previously (too much space):
[image]

With this PR:
[image]
[image]

Optionally all current individual toolbar buttons for the 7 views are still available:
[image]
[image]
closed 2024-03-18 by sliptonic