FreeCAD/FreeCAD #12867: Export filename preset is empty

m0n5t3r, 2024-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I suspect this is a recent breakage - it used to be that the exported filename (when selecting a body and pressing `ctrl+e`, for instance) would be preset to `<fcstd-file>-<body-name>`, and the selected output format would add the extension automatically. Right now, the filename field is always empty, and it seems to require the whole name including extension to be typed, every time...

Adding the settings keys mentioned at https://wiki.freecad.org/Fine-tuning#Default_export_filename doesn't seem to help.

[image]
[image]

### Full version info

```shell
OS: Ubuntu Core 22 (Budgie:GNOME/budgie-desktop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36340 (Git) Snap 952
Build type: Release
Branch: main
Hash: addf08ce6b0df264a47e836e799b6756626f146e
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.5.14
  * A2plus 0.4.64
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-11:
I cannot reproduce, its always populated. On stable 0.21.2 and on
<details><summary>Windows 11 0.22dev</summary>
<p>

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2

``` 

</p>
</details>

m0n5t3r, 2024-03-11:
I managed to trace it to `BaseApp/Preferences/Dialog/DontUseNativeDialog`, which was set to `false`; no idea when that preference showed up, it's not present after starting without `user.cfg`  (so forcing default values).

The option also causes this behaviour in stable, btw, this is how I figured it out (saw stable doing the same thing, reset profile directory, saw it worked, loaded the configurations in vimdiff and came across that param group (which also includes `DontUseNativeFontDialog`, set to `true`).

m0n5t3r, 2024-03-11:
It seems that configuration option has been added 6 years ago; it's there on my old computer, and freecad-daily works with the native gtk dialog without issues; there are 2 major differences between the computers: ubuntu release (old one is 22.04, this one is 23.10) and deb (old system) vs snap (this system).

Going to try installing the deb tomorrow to see how it does, since the stable appimage won't start (hitting https://github.com/FreeCAD/FreeCAD-Bundle/issues/67).

m0n5t3r, 2024-03-14:
one more data point, since I figured out where the weekly appimages were (those don't crash with radeon drivers): the weekly appimage seems to simply ignore the parameter and uses the qt file dialog instead of the system one, so the name is filled in correctly.

<details>
<summary>
FreeCAD_weekly-builds-36391-2024-03-11-conda-Linux-x86_64-py310.AppImage
</summary>

```
OS: Ubuntu 23.10 (Budgie:GNOME/budgie-desktop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git) AppImage
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.6.24
  * OpenDark 2023.12.17
  * btl 0.9.9
  * kicadStepUpMod 11.1.0
  * sheetmetal 0.4.3
  * CurvedShapes 1.0.5
  * woodworking 0.21.2.33771
```
</details>

If using the native dialogs is in process of being deprecated, I suppose this can be closed (and hopefully help whoever may come across this - removing the parameters mentioned above "fixes" the problem by reverting to the qt file dialog)

m0n5t3r, 2024-03-14:
> So not reproducible with the latest weekly builds? Can I close this issue?

reproducible with a tight set of constraints (snap, gnome-based gtk environment, native file dialogs); I asked in the forum if it's a corner case worth pursuing, or native dialogs are being retired anyway

-------- Original Message --------

plyblu, 2024-06-11:
I'm using  0.21.2 on Win10, came across a change in FC behavior, and found this thread when searching for info. Previous behavior for me was that it used  `<fcstd-file>-<feature-name>` until I typed in a filename. Then it used the filename until FC was stopped. On FC restart, it reverted back to `<fcstd-file>-<feature-name>`. For some unknown reason, it's now remembering the typed-in filename even after a restart. Also, for me it has always used `<feature-name>`, not `<body-name>`. Perhaps the other commenter selects the body when exporting but I learned early on that only works sometimes whereas selecting the last feature in the body always works (using PartDesign).

IMHO the export filename default has some idiosyncrasies that could be improved upon. Currently I'm not sure how to get my install back to using `<fcstd-file>-<feature-name>` as the default filename. And I've never found the `<feature-name>` part of the default filename useful since it's typically a dress-up feature like a fillet or chamfer.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```

FoobarTheFatPenguin, 2024-09-04:
I can confirm the behaviour on Linux Mint 21.3 using the Flatpak version of FreeCAD (0.21.2).

The export filename is missing. The aforementioned settings parameter (General/ExportDefaultFilenameSingle) was also not present in the config. Manually adding it had no effect. The filename is always empty when exporting.

But if I set Dialog/DontUseNativeDialog to true, then the file name is pre-filled as expected.

luzpaz, 2024-09-04:
@FoobarTheFatPenguin can you reproduce on development build ? (flatpak excluded)  

https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds  
&  
https://wiki.freecad.org/Ubuntu_Snap#Changing_between_different_Snaps

FoobarTheFatPenguin, 2024-09-04:
No, it does seem to work with that version (using FreeCAD_weekly-builds-38622-conda-Linux-x86_64-py311.AppImage) from the weekly builds page.

m0n5t3r, 2024-09-04:
I switched to the `latest/edge` snap (I'm usually on beta), it does the same thing

```
OS: Ubuntu Core 22 (
Budgie:GNOME/budgie-desktop
)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38635 (Git) Snap 1125
Build type: Release
Branch: main
Hash: 7274dac185339a45020f1dc91f2b6c7478b2b0aa
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods:
  * sheetmetal.backup1711992092.1663022 0.4.5 (Disabled)
  * Curves 0.6.29
  * OpenDark 2023.12.17
  * btl 0.9.9
  * kicadStepUpMod 11.1.0
  * sheetmetal 0.4.10
  * freecad.gears 1.2.0
  * CurvedShapes 1.0.8
  * woodworking 0.21.2.33771
```

luzpaz, 2024-09-04:
@FoobarTheFatPenguin please always add the full [About](https://wiki.freecad.org/About) info of any build you test.  
So you can also reproduce on the AppImage as well ?

FoobarTheFatPenguin, 2024-09-04:
The AppImage works fine and does not have the same problem.

The Flatpak version that has the issue:
```
OS: KDE Flatpak runtime (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * freecad.gears 1.0.0
  * fasteners 0.4.71
  * ProDarkThemePreferencePack 1.0.0
  * Behave-Dark-Colors 0.1.1
  * Dracula 0.0.5
  * ArchTextures
```

The AppImage weekly build that does NOT have the problem:
```
OS: Linux Mint 21.3 (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
```

maxwxyz, 2024-12-15:
Is this still an issue in 1.0?

FoobarTheFatPenguin, 2024-12-16:
> Is this still an issue in 1.0?

I'm unable to reproduce it with the 1.0.0 flatpak package. It seems to work - for now.

closed 2024-12-16 by maxwxyz