FreeCAD/FreeCAD #12869: Objects are being added on top and also in the wrong order of the Objects tree

leoheck, 2024-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This looks like a regression. When objects are created, for instance with Part Design Clone tool, objects are been added on top of the existing object, in the Objects tree. Also, When you move objects let's say out of a folder/part and in again to organize/set the order, the items are not being added in the same order they were selected.

### Full version info

```shell
OS: Ubuntu 23.10 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36331 (Git) AppImage
Build type: Release
Branch: main
Hash: eb5a1b333946ce4ac8a4ead835cba8bdcf602a87
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * POV-Ray-Rendering
  * Assembly4 0.50.7
  * sheetmetal 0.4.2
  * InventorLoader 1.4.0
  * ose-piping
  * fasteners 0.5.13
  * OpenDark 2023.12.17
  * ProDarkThemePreferencePack 1.0.0
  * Render 2024.1.23
  * OpticsWorkbench 1.0.11
  * lattice2 1.0.0
  * kicadStepUpMod 10.23.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leoheck, 2024-03-11:
Hmm, I checked this option last week. Could you try it again to enable this?
[image]

maxwxyz, 2024-03-11:
This setting also makes no difference on my side. Copies and new objects are added at the bottom of the tree.

leoheck, 2024-03-11:
Interesting... I am testing this with a different design. And I could not reproduce it too. I will close until I can reproduce this.

closed 2024-03-11 by leoheck