FreeCAD/FreeCAD #12870: FEM examples in the Start wb should be updated

FEA-eng, 2024-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FEM examples shown in the Start wb should be updated to account for the recent changes - removal of the CalculiX new framework solver, addition of 2D analyses, change in result units system (in 0.21)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-11:
@FEA-eng it would be great if a tutorial or description inside the file could be supplied. Not sure if this is possible but this also came up during the FreeCAD Day 20204 Complaint Session.

FEA-eng, 2024-03-11:
> @FEA-eng it would be great if a tutorial or description inside the file could be supplied. Not sure if this is possible but this also came up during the FreeCAD Day 20204 Complaint Session.

Sure, I'll think about this next. FEM examples (not the ones on the Start page but the ones in this library: [FEM Examples](https://wiki.freecad.org/FEM_Examples)) already have such [text documents](https://wiki.freecad.org/Std_TextDocument) called Explanation_Report. But maybe they should be added for other Start wb examples too (especially the Assembly one).

PR #12871 (2024-03-15): Start wb FEM examples update
fixes #12870 but also resolves #5701

- removes cantilever new solver example
- updates old cantilever 2D and 3D examples
- adds new cantilever 1D example

@marioalexis84 FYI
closed 2024-03-15 by chennes