FreeCAD/FreeCAD #12875: Correction of non-zero Z nodal coordinates for 2D CalculiX FEM analyses

FEA-eng, 2024-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Recently, 2D analyses with CalculiX were implemented. In some rather uncommon cases, the meshes might be generated with non-zero Z nodal coordinates (having very low values like e-12) causing CalculiX error. 

The first question is when it may happen. For example, I've noticed that it happens when Part Plane is created on another global CSYS plane and then rotated to the XY plane using the Transform manipulator. Probably due to some small misalignment. But it doesn't happen when the same Part Plane is created already on the XY plane.
Here's a report from before 2D analyses where implemented and a user configured them manually (by editing the input files): https://forum.freecad.org/viewtopic.php?t=81582#p710993 His geometry was quite complex and I don't know what operations led to this issue. Maybe it's because he used A2plus constraints to align the parts.

Now some ideas on how we could prevent this:
1. use some countermeasures offered by Gmsh or Netgen - I don't know if they are available at all and it's rather unlikely
2. somehow force the geometry to lie exactly on the XY plane - are there any tools that could help? It seems that [Draft Shape2DView](https://wiki.freecad.org/Draft_Shape2DView) can serve as a workaround but then the faces have to be recreated from wires using [Draft Upgrade](https://wiki.freecad.org/Draft_Upgrade)
3. search for and correct those coordinates (replace them with zeros) after the mesh in the .inp file is generated - this could be automatically done for each 2D analysis run but would probably result in longer time to write the final .inp file
4. don't write the Z (third) nodal coordinates for 2D analyses at all - it seems that CalculiX allows this but it could potentially be troublesome in other ways

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marioalexis84, 2024-04-10:
@FEA-eng   do you have another example? I can't reproduce with the forum file.

marioalexis84, 2024-04-10:
Even CalculiX is too restrictive in requiring z-coordinate=0 since it is reasonable to have a planar shape outside the XY plane.

For a generic planar shape what we can do is internally rotate the points, project in the XY plane (to eliminate floating point errors) and after the simulation transform the results with the inverse placement.

marioalexis84, 2024-06-11:
@FEA-eng the example [assembly.zip](https://github.com/FreeCAD/FreeCAD/files/14931946/assembly.zip) fails even using all z coordinates equal to 0.

PR #14619 (2024-06-17): Fem: Force z=0 for plane stress, plane strain and axisymmetric elements in writeABAQUS - fixes #12875
Fixes #12875 

@FEA-eng
commit f8027a16 (2024-06-17): Fem: Force z=0 for plane stress, plane strain and axisymmetric elements in writeABAQUS - fixes #12875
closed 2024-06-17 by chennes