FreeCAD/FreeCAD #12880: Attaching sketch to datum plane results in error

komoricodrutz, 2024-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi,
As described above.
Working in Part Design, I created a simple extrude of a sketch. Then placed a datum plane at the top of the previous extrusion. When creating a sketch on the respective datum plane, it results in an error:
PositionBySupport: Null face in AttachEngine3D::calculateAttachedPlacement()!
When attempting to constrain the contents of the new sketch, it randomly turns to orange, redundantly constrained.
If I attach the sketch directly to the object face, it suddenly works without a hitch.
Wherever the datum plane is placed relative to the already existing extruded object, the error will be the same.
I have been using this practice for years now, to avoid the known topo-naming issue.

This happens both when editing an existing file and when creating a new one from scratch.
The screenshot below shows it. The erroring sketch should contain a centered circle, however, as you can notice, it is not even displayed on top of the plane.

Thank you!

### Full version info

```shell
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36338 (Git)
Build type: Release
Branch: main
Hash: 5b5be480a3a14963cbdefe2e7a3923f9136af6a9
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BOLTSFC 2022.11.5
  * A2plus 0.4.60n
  * Lithophane
  * Assembly4 0.50.2
  * Glass
  * fasteners 0.4.62
  * fcgear 1.0.0
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-03-11:
Looks to me that the fix has already been submitted https://github.com/FreeCAD/FreeCAD/pull/12861

maxwxyz, 2024-03-11:
I cannot reproduce on Windows 11 0.22dev
<details><summary>Details</summary>
<p>

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35758 (Git)
Build type: Release
Branch: sketcher-offset-external-geometry
Hash: 7e8265fb10269566c01bf11d8b1600a5dd81aaa2
Python 3.11.7, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.24
  * Defeaturing 1.2.2
  * fasteners 0.5.13
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

</p>
</details>

komoricodrutz, 2024-03-11:
Hi, both.
> I cannot reproduce on Windows 11 0.22dev
> Details

That's an older build. (35758). I didn't have this issue before yesterday.
The previously mentioned one is **36338**.
I've just received the update to 36340. Issue still persists.
Using the freecad/daily ppa.

> Looks to me that the fix has already been submitted #12861

There is no crash happening. Just the error thrown out and the sketch is basically unusuable. If I attach it to a face on the object or straight to the base plane and then add a z offset, it works. The behaviour only occurs when attaching sketch to a datum plane.

New Details:
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36340 (Git)
Build type: Release
Branch: main
Hash: addf08ce6b0df264a47e836e799b6756626f146e
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BOLTSFC 2022.11.5
  * A2plus 0.4.60n
  * Lithophane
  * Assembly4 0.50.2
  * Glass
  * fasteners 0.4.62
  * fcgear 1.0.0

maxwxyz, 2024-03-12:
[image]
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.24
  * Defeaturing 1.2.2
  * fasteners 0.5.13
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2

```

komoricodrutz, 2024-03-12:
Just updated to latest ppa version.
Issue persists.

OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36392 (Git)
Build type: Release
Branch: main
Hash: 9140327832b94ea4325dfcf185e2e68659875e47
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BOLTSFC 2022.11.5
  * A2plus 0.4.60n
  * Lithophane
  * Assembly4 0.50.2
  * Glass
  * fasteners 0.4.62
  * fcgear 1.0.0

komoricodrutz, 2024-03-22:
Hi,
I've been away for about a week. The sketch-to-plane attachment has apparently been solved in the mean time. Updated to this version:
<details>
<summary>Build Details</summary>

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36500 (Git)
Build type: Release
Branch: main
Hash: 161b0fb6ee9418473bc19b36976d3f06c4ca098c
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BOLTSFC 2022.11.5
  * A2plus 0.4.60n
  * Lithophane
  * Assembly4 0.50.2
  * Glass
  * fasteners 0.4.62
  * fcgear 1.0.0

```

</details>

However, there now seems to be another issue. Unknown if related:
If I FIRST select a base plane and THEN try to attach a Datum Plane to it, I am getting an error:
[image]

If, however, I create the Datum Plane and then select the reference plane, or if after OK-ing that error message I manually re-select the reference base plane, it works:

https://github.com/FreeCAD/FreeCAD/assets/134259571/36258ffc-efbc-44dd-b08e-fa5e81eba638

komoricodrutz, 2024-03-25:
The previously mentioned additional bug seems fixed in the latest version 0.22.0dev.36534 (Git):

<details>
<summary>Build Details</summary>

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36534 (Git)
Build type: Release
Branch: main
Hash: d5b90e50af6a758af748179f289bb8f09e357266
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BOLTSFC 2022.11.5
  * A2plus 0.4.60n
  * Lithophane
  * Assembly4 0.50.2
  * Glass
  * fasteners 0.4.62
  * fcgear 1.0.0

```

</details>

Guess this bug can be considered closed.
Thank you!

closed 2024-03-25 by maxwxyz

komoricodrutz, 2024-03-26:
Both issues seem to have now been solved, so I guess the bug can now be closed.

<details>
<summary>Build Details</summary>

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36535 (Git)
Build type: Release
Branch: main
Hash: e5d4f0e38eb3a0c867d5ef383db081285ffeeba1
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BOLTSFC 2022.11.5
  * A2plus 0.4.60n
  * Lithophane
  * Assembly4 0.50.2
  * Glass
  * fasteners 0.4.62
  * fcgear 1.0.0

```

</details>