FreeCAD/FreeCAD #12881: Path: Mill -> Rectangular part machines as parallelogram

mn00n, 2024-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

 Gcode for a rectangular G1 path doesn't exactly define the X-axis as horizontal (or Y-axis as vertical).  The error appears to occur when a custom cutter diameter is used.

[image]

chrisb provided this response in the forum:
I can confirm the issue. I created a new job in 0.22 and it it behaves the same. The issue doesn't occur with the default 5mm tool, but with the 2mm tool it is off by nearly 9/1000. The image shows between the mark the start and end X values:

[image]

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
[/code]
```

### Subproject(s) affected?

Path

### Anything else?

[Pad_test.zip](https://github.com/FreeCAD/FreeCAD/files/14561391/Pad_test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2024-04-02:
Welcome to the world of floating point math.  This isn't solvable.  
The default geometry and curve accuracy is 10 micron which is good enough for most uses.  You can increase your accuracy to 1 micron and you'll be generating gcode too accurate for pretty much any CNC machine on the planet.  It still won't have nice round number in some cases though.

closed 2024-04-02 by sliptonic

mn00n, 2024-04-02:
Appreciate the sanity check.
After troubleshooting further it appears backlash compensation in Mach3 plays a part in blemishes I'm seeing.  Turning it off results in smooth edges (using the same gcode).