FreeCAD/FreeCAD #12883: Assembly: Multiple joints conflict and move grounded part

maxwxyz, 2024-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. create an assembly with 2 objects, ground one part
2. create a distance constraint with an offset between two surfaces
3. create a fixed constraint between the same two surfaces
4. move around the assembly

The assembly can be moved, there are a few issues that it cannot solve and then the grounded part moves.
After deletion of the distance and the fixed joint sometimes the grounded part still flips or moves.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35758 (Git)
Build type: Release
Branch: sketcher-offset-external-geometry
Hash: 7e8265fb10269566c01bf11d8b1600a5dd81aaa2
Python 3.11.7, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.24
  * Defeaturing 1.2.2
  * fasteners 0.5.13
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-11:
@PaddleStroke

PaddleStroke, 2024-03-12:
That's on my dev branch?

maxwxyz, 2024-03-12:
@PaddleStroke yes, PR #12406

PaddleStroke, 2024-03-12:
You are purposefully making conflicting constraints. The solver will try to remove redundant joints. But yes I'm not surprised it's buggy. Not sure yet how to handle conflicting joints.

dustinlacewell, 2024-04-16:
I'm attempting to use a fixed joint between two planes to create a coplanar constraint. Then I am trying to position the second along the plane using distance joints between edges. Conceptually, these joints shouldn't conflict, but I experience the behavior of random solves and base object moving around.

dustinlacewell, 2024-04-16:
Nevermind, I understand that the fixed joint really is a fixed joint, so we're overconstraining the assembly here.

PaddleStroke, 2024-06-06:
This issue should perhaps be closed since 14528 is opened to ask for conflict detection.

closed 2024-10-30 by maxwxyz