FreeCAD/FreeCAD #12884: Assembly: Click in the 3D view while editing joint clears it

maxwxyz, 2024-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When editing a joint an accidental click in the 3D view (empty space) will delete all linked JCS / elements in the joint.
Steps to reproduce:
1. create an assembly with joints
2. edit a joint
3. click in the 3D view empty space. (click on geometry is prohibited but not in the empty space)
4. joint elements are deleted

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35758 (Git)
Build type: Release
Branch: sketcher-offset-external-geometry
Hash: 7e8265fb10269566c01bf11d8b1600a5dd81aaa2
Python 3.11.7, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.24
  * Defeaturing 1.2.2
  * fasteners 0.5.13
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-11:
@PaddleStroke

furgo16, 2024-03-11:
I can confirm this. I noticed this a few days ago and I was a bit puzzled, I thought it was me doing something wrong.

PaddleStroke, 2024-03-12:
This is because clicking on empty space is clearing the selection. And as the tool is using the selection, if you click aside it clears the selection.
I can change to greedy selection in this tool to remove this issue.

maxwxyz, 2024-03-12:
@PaddleStroke how could you otherwise delete one selection? I can select it in the task dialog and hit delete but I also cannot remove with right click. Maybe the selection should also be visible when the joint is open in the task view.

PaddleStroke, 2024-03-12:
Erratum, we are already in greedy selection for joint edition.
I don't know if we can prevent selection clearing on click. I will try some things.

maxwxyz, 2024-03-12:
Dialog would be a good solution. Or a double click in empty space to clear selection.

PaddleStroke, 2024-03-12:
> There is the way to delete an item from the task panel but this hides the linked objects and the panel is stuck in a weird state. When closing it then it's not even possible to re-open it by double-clicking the Joints object.

This bug is fixed in current dev branch. So you can delete in the list and it unselect the elements.

> And what if the click is not by accident but on purpose? AFAICS clearing the selection on purpose is the only way to select two different shape elements.

Note that this would be only temporary, only while the joint creation tool is active. This would actually be the same behavior as the PartDesign fillet tool. When you have something selected in PD fillet, clicking in void do not unselect it. To unselect you have to click again on the element or deleting in the list.

PaddleStroke, 2024-03-13:
So looking into it, it makes sense that when using greedy selection, clicking empty space do not clear selection.
This behavior was actually first implemented in 2017 https://forum.freecad.org/viewtopic.php?style=1&t=24215&start=10
When Ctrl was held. @wwmayer you implemented it in https://github.com/FreeCAD/FreeCAD/commit/98470fe

But then RT made a commit that broke it I think. https://github.com/FreeCAD/FreeCAD/commit/c744157e9a281186a1af086e637edaf552875911#diff-678eb0ebc349cdbfef8b0b98ceb5b263c6894b1336662d6e0f5167e384ae468f

Then it got fixed somewhere apparantly because holding Ctrl prevent clearing of selection (while greedy sel does not).
So basically before it was implemented in this : 
```
            bool greedySel = Gui::Selection().getSelectionStyle() == Gui::SelectionSingleton::SelectionStyle::GreedySelection;
            if(setSelection(infos, event->wasCtrlDown() || greedySel))
                action->setHandled();
```

But now `setSelection `returns false when `infos `is empty. So the prevention of clearing on Ctrl-holding must be done somewhere else.
What about modifying setSelection so that it acts as it was before? Like so : 

```
bool SoFCUnifiedSelection::setSelection(const std::vector<PickedInfo> &infos, bool ctrlDown) {
    if (infos.empty()) {
        return ctrlDown;
    }
    if (!infos[0].vpd) {
        return false;
    }
```

Is that acceptable @wwmayer ?

wwmayer, 2024-03-13:
> But then RT made a commit that broke it I think. 

Obviously yes.

> What about modifying setSelection so that it acts as it was before? Like so :

Why not going further and return ctrlDown everywhere where it returns false because this was the old behaviour?

PaddleStroke, 2024-03-13:
> Why not going further and return ctrlDown everywhere where it returns false because this was the old behaviour?

Got it I'll do that then

wwmayer, 2024-03-13:
Or you can write it as:
```
            if (setSelection(infos, event->wasCtrlDown()) || event->wasCtrlDown() || greedySel)
                action->setHandled();
```
because then we don't depend on implementation details of setSelection()

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
closed 2024-05-06 by yorikvanhavre