FreeCAD/FreeCAD #12887: Assembly: Fixed joint offset/rotation moves grounden part

maxwxyz, 2024-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a fixed joint and changing offset and rotation, the grounded part moves, instead of the *loose* one.
When flipping, sometimes it resets or all part move.
Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/8cc85d8a-9051-41b2-b91e-3d555562183c

File:
[testassy.zip](https://github.com/FreeCAD/FreeCAD/files/14563408/testassy.zip)

Tested with #12406 

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35758 (Git)
Build type: Release
Branch: sketcher-offset-external-geometry
Hash: 7e8265fb10269566c01bf11d8b1600a5dd81aaa2
Python 3.11.7, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.24
  * Defeaturing 1.2.2
  * fasteners 0.5.13
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-11:
@PaddleStroke 
It also moves when on some selection edged, at the end FreeCAD crashes:

https://github.com/FreeCAD/FreeCAD/assets/6246609/95e325f0-e8dd-4c9e-9d34-b825413b4728

PaddleStroke, 2024-03-12:
This is in the dev branch 12406?

maxwxyz, 2024-03-12:
yes

maxwxyz, 2024-03-20:
@PaddleStroke not sure if related but when using a fixed joint and increasing or decreasing the rotation it always rotates in the same direction and when clicking OK its in a different position. Try with scrolling in the rotation boy of the fixed joint.

PaddleStroke, 2024-03-20:
I cannot reproduce this rotation issue. Can you provide a file please?

maxwxyz, 2024-03-25:
I also cannot reproduce. Maybe I've tried too many different things and broke the assembly before.

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
closed 2024-05-06 by yorikvanhavre