FreeCAD/FreeCAD #12889: Assembly: Deleting a part leaves joint. Parts cannot be moved.

maxwxyz, 2024-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When deleting a part which is part of a joint the joint is left alive. You cannot double click or edit that joint. You have to manually delete it. Whether you delete this joint or leave it, the parts of the entire assembly cannot be moved manually by dragging until you restart FreeCAD
Demo file with a leftover joint (will work as you open it fresh). Just create a new joint and delete the part. 
If you create a distance joint with the cylinder (grounded) and the sphere and then delete the sphere you can move the grounded part again.
Tested with #12406 

Demo:

[deleted-items.zip](https://github.com/FreeCAD/FreeCAD/files/14563530/deleted-items.zip)

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35758 (Git)
Build type: Release
Branch: sketcher-offset-external-geometry
Hash: 7e8265fb10269566c01bf11d8b1600a5dd81aaa2
Python 3.11.7, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.24
  * Defeaturing 1.2.2
  * fasteners 0.5.13
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-11:
@PaddleStroke

PaddleStroke, 2024-03-12:
Ok so first thing is that I added a failsafe to filter out incomplete joints. On top of that I am disabling joints when one of their linked part disappear.
However I cannot find how to delete the joint when the link disappear.

I tried to do : `joint.Document.removeObject` but it crashes freecad. I guess after the proxy onChanged, the joint object is accessed. And as it's been deleted it crashes.
```
    def onChanged(self, joint, prop):
        """Do something when a property has changed"""
        # App.Console.PrintMessage("Change property: " + str(prop) + "\n")
        if prop == "Part1" or prop == "Part2":
            if activeTask:
                # if joint is being edited then it's ok
                return

            # Check that the object still exist. This is to catch when user 
            # deleted one of the parts.
            if joint.Part1 is None or joint.Part2 is None:
                joint.Activated = False
                #joint.Document.removeObject(joint.Name)
                return
```

@maxwxyz do you think it's enough if the joint is disabled?
@wwmayer do you know how I could delete the joint object from the onChanged? Is there something like 'delete later'?

maxwxyz, 2024-03-12:
@PaddleStroke yes could be, if the disabled is displayed somehow in the tree. Otherwise we have many joints in the tree and you don't know which one is disabled. What happens when I try to edit a disabled joint and add a new link to it, is it then automatically enabled again?

PaddleStroke, 2024-03-12:
@maxwxyz yes I agree it could polute the jointgroup very quickly. Let's see if Werner has some knowledge on this :)
If you edit the disabled joint and add a new link, it does not activate by itself. So perhaps de-activating it is not so smart. We can just ignore incomplete joints.

wwmayer, 2024-03-12:
> do you know how I could delete the joint object from the onChanged?

That's not possible because the pointer of the calling instance that calls onChanged will become dangling and there is no good way to avoid this (-- unless all raw pointers are replaced with shared and weak pointers).

> Is there something like 'delete later'?

You can implement a way to postpone the deletion. All what you need is to store the name of the objects that should be deleted and then perform it at a later point.

PaddleStroke, 2024-03-12:
Thanks for the feedback.
Ok then I made it so that the incomplete joints are deleted when getJoints is called (ie when solving).

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
closed 2024-05-06 by yorikvanhavre