FreeCAD/FreeCAD #12890: Assembly: Cannot find parts in open documents

maxwxyz, 2024-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I want to insert parts from another (already opened) file. 
It is not possible, it does not show any parts, maybe that is because all bodies are in an assembly in this other file? 
I've tested this with #12406 so #12440 should be fixed.
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35758 (Git)
Build type: Release
Branch: sketcher-offset-external-geometry
Hash: 7e8265fb10269566c01bf11d8b1600a5dd81aaa2
Python 3.11.7, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.24
  * Defeaturing 1.2.2
  * fasteners 0.5.13
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-11:
@PaddleStroke

PaddleStroke, 2024-03-12:
So I see 2 issues here : 
A - Objects within an assembly does not show in the list. Even if it is not the active assembly (which was the intendion)
B - Assemblies do not appear in the insert tool. So the problem here is that I added support for assemblies objects, but not for LINKS to assembly objects (which is what the insert tool would make).
The problem with links to assemblies, is that changing the objects placements of the linked assembly, means changing placement of the objects in the original assembly. Which could be acceptable. But what happens if we have 2 links to the same assembly? Then both links are showing the same objects. So they cannot move independantly.

pierreporte, 2024-03-12:
> The problem with links to assemblies, is that changing the objects placements of the linked assembly, means changing placement of the objects in the original assembly. Which could be acceptable. But [...]

Moving sub-assemblies from a parent assembly shouldn’t modify the sub-assemblies. When a sub-assembly is moved, it is the origin of the assembly that moves, but components in it are still defined from that origin so they move with it.

There is the discussion in #12440 about static and rigid assemblies. Two instances of the same sub-assemblies, when flexible, should be able to freely move independently. A pneumatic cylinder, for example, when used multiple times, can be fully compressed or fully extended depending on where it is. But when opened independently, it should show its default configuration which can be different from all existing instances.

Of course, when an sub-assembly is edited from its parent assembly, all the instances need to update accordingly. For flexible assemblies, I don’t know what should be displayed though.

PaddleStroke, 2024-03-13:
I have reworked the Insert tool. It now uses a tree : 
[image]

Can you test it out in #12406?

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
closed 2024-05-06 by yorikvanhavre