FreeCAD/FreeCAD #12892: Arch Space should include more IFC Attributes as variables for text to show

oscarale-camorlinga, 2024-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Arch Space tool has several variables that can be referenced within the Text attribute such as `$area`, `$label`, `$tag`, etc. I think it would be proper to also include the `Name`, `Long Name` and `Description` IFC attributes. According to the [IFC4 Documentation](https://standards.buildingsmart.org/IFC/RELEASE/IFC4_1/FINAL/HTML/schema/ifcproductextension/lexical/ifcspace.htm) the IfcSpace entity should include the following attributes:

- _Name_ holds the unique name (or space number) from the plan.
- _Description_ holds any additional information field the user may have specified, there are no further recommendations.
- _LongName_ holds the full name of the space, it is often used in addition to the Name, if a number is assigned to the room, then the descriptive name is exchanged as LongName.
- _ObjectType_ holds the space type, i.e. usually the functional category of the space

It would be helpful to display some of this information for spaces, especially when you want to have the Label of the Arch Space object different from the actual name being shown in the text. The variables could be for example `$name`, `$longname` and `$description`. Looking at [ArchSpace.py](src/Mod/Arch/ArchSpace.py) this seems like a simple enough fix which I, a novice in Python, could even make a PR for, though I would need to do some more investigation on how to add the `Name` IFC Attribute to the Arch Space object.

Is this something that could be implemented?

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * BIM 2021.12.0
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-03-12:
All these are already exposed:

[image]

Only specific thing to know is that in FreeCAD, the Name attribute refers to the internal, unmutable FreeCAD object name. The user-defined, modifiable name is the "Label" property.

oscarale-camorlinga, 2024-03-12:
> Only specific thing to know is that in FreeCAD, the Name attribute refers to the internal, unmutable FreeCAD object name. The user-defined, modifiable name is the "Label" property.

I see, so Name couldn't be used as the IFC Attribute. Then perhaps 'Space Number'? Seeing as it references that in the IFC4 spec 'Name' denotes a _unique name (or space number) from the plan_.

yorikvanhavre, 2024-03-12:
Oh sorry, I didn't understand your query. Yes, that would be very easy to do. Basically it happens here, one would just need to extend the list here: https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Arch/ArchSpace.py#L651

Would you like to try? I can explain how to do it, and you'd become a FreeCAD contributor :innocent:

yorikvanhavre, 2024-03-13:
Yes, exactly!

oscarale-camorlinga, 2024-03-19:
I was having issues compiling, but I got it figured out now.

[image]

PR #13039

Thanks!

closed 2024-03-20 by yorikvanhavre