FreeCAD/FreeCAD #12901: TechDraw: Scale of views of ProjGroup are not correct, if the Page is scaled when the ProjGroup is created.

edi271, 2024-03-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Scale of views of **ProjGroup** are not correct, if the **Page** is scaled when the **ProjGroup** is created.

- Create a new page.
- Set scale of the page to a value other then 1.0.
- Create a projection group. Keep **Page** as scale type. Press **OK**.

The created **ProjGroup** has the scale of the page. Correct.
The created view **Front** has the scale 1.0. Not correct.

### Full version info

```shell
OS: Ubuntu 22.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36350 +1 (Git)
Build type: Unknown
Branch: IssueAxoLengthDimension
Hash: 734aea5cf6f901c7502ec658d2b399d015795a44
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Austria (de_AT)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-12:
I cannot reproduce on 
<details><summary>Windows 11 0.22dev</summary>
<p>

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.24
  * Defeaturing 1.2.2
  * fasteners 0.5.13
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2

``` 

</p>
</details> 

The front view itself has no scale parameter but the projection group is set to the same scale as the page.

edi271, 2024-03-12:
The front view has a scale parameter, but it is hidden in the Data panel of the combo view.  The front view is of type TechDraw::DrawProjGroupItem and has a property Scale. Send the front view to the python console, and have a look to its Scale property.
The AxoLengthDimension tool of TechDraw uses this property, and fails because it receives the wrong scale value 1.0 instead of the the scale value of the parent page.

maxwxyz, 2024-03-12:
Yes, I can reproduce it now. This is also happening, if you change the page scale later. I cannot overwrite the individual view scale.
@WandererFan

WandererFan, 2024-03-12:
The ProjectionGroupItems don't use their own scale property (which is why it is hidden and read-only). 

They call getScale() (as do all the other view types).  getScale sorts out whether an object should be using it's own scale or a parent's scale and handles all the Page/Custom/Auto stuff.

I don't think I've ever got around to exposing getScale to Python, but I will check.

maxwxyz, 2024-03-12:
Ok, but the axonometric dimension tool uses the wrong scale then as described in edis comment.
@edi271 is it only this tool? Could you rephrase the issue?

edi271, 2024-03-13:
@WandererFan: If you agree, I will try to expose getScale() to Python.

edi271, 2024-03-13:
The getScale() function has been exposed to Python by WandererFan.

closed 2024-03-13 by edi271