FreeCAD/FreeCAD #12913: The Suppressed property is not accessible for users

NomAnor, 2024-03-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Suppressed property introduced in #12096 is marked hidden and is not (easily) accessible to users.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36338 +18 (Git)
Build type: Debug
Branch: integration
Hash: 65582efdd0d9407951a6fc996cb36662efb7b2c9
Python 3.11.8, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.3.0
Locale: German/Germany (de_DE)
Installed mods: 
  * Manipulator 1.5.7
  * ThreadProfile 1.84.0
  * DynamicData 2.60.0
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.0.0
  * Assembly3 0.11.3
  * QuickMeasure 2022.10.28
  * fasteners 0.4.21
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-03-13:
Related issue https://github.com/FreeCAD/FreeCAD/issues/11212. Possibly a duplicate.

NomAnor, 2024-03-13:
You're right, I will close this.

closed 2024-03-13 by NomAnor