FreeCAD/FreeCAD #12915: DirectWrite: CreateFontFaceFromHDC() failed

Manuelzwei, 2024-03-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Error after new installation, when starting, before opening an existing file:
[image]

Where is the 8514oem font located on my PC? 

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Luxembourg (de_LU)
Installed mods: 
  * kicadStepUpMod 10.17.1
  * offline-documentation 1.0.0-alpha
  * sheetmetal 0.3.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Manuelzwei, 2024-03-13:
Thank you for reply.
I see this error by an upgrade. So I deinstall FreeCAD, reboot and install it new. The same error was shown.

Manuelzwei, 2024-03-13:
FreeCAD 0.22.0dev: same situation.

I'm getting a new Windows 11 PC this weekend. Then I reinstall everything and try it again

luzpaz, 2024-03-14:
Related to #10555 ?

Manuelzwei, 2024-03-14:
> Related to #10555 ?

Yes, I have 125% DPI. So I think it was the same as #10555.
I to set my DPI to 100% and it was all right.

Thank you.

closed 2024-03-14 by Manuelzwei