FreeCAD/FreeCAD #12916: The windows cannot be moved across multiple screens

Manuelzwei, 2024-03-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

On the left side the arrangement of the screens used and their resolution is shown (Screen 1).
Screen 2 + 3 shows FreeCAD.
The windows 'Page', 'Spreadsheet, ...' cannot be moved to screens 4 + 5.
However, the status messages are moved to screens 4 and 5.

I aligned the screens exactly (seen by the straight line between screen 2 + 3).

Screen 4 should be a little higher/upper - the display has been moved slightly for demonstration purposes.

Screens 2, 3, 4 and 5 have the same resolution.
Screen 1 have a smaller one. 
Moving the windows also doesn't work if screen 1 is not present.

If the screens 2 + 3 are not aligned to the nearest pixel, the windows can be selected, but no reasonable actions can be carried out.
Among other things, it is then not possible to select the menu items File, Edit, Views and so on.

What to set to place the windows independently?

Thank you!

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Luxembourg (de_LU)
Installed mods: 
  * kicadStepUpMod 10.17.1
  * offline-documentation 1.0.0-alpha
  * sheetmetal 0.3.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-03-13:
@Manuelzwei This may be a Qt issue. Can you try to reproduce on development version ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Manuelzwei, 2024-03-13:
> @Manuelzwei This may be a Qt issue. Can you try to reproduce on development version ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Thank you!
What is Qt?

Now I'm using FreeCAD 0.22.0dev.
Same problem – and another one, I think.
Now I have the main FC screen stretched across half of the top screens and the entire bottom two screens.
The TechDraw and Spreadsheet windows still cannot be removed from the main screen.
Additionally, when I try to zoom in on the Part Design model with the mouse, the other open windows, TechDraw and Spreadsheet, suddenly change.

Manuelzwei, 2024-03-13:
It's surprising that the combo box and other windows can be freely placed on the screens, but the work windows cannot.

luzpaz, 2024-08-25:
@Manuelzwei can you reproduce on latest 0.22 dev ? Please add the full About info when you test. TIA!

closed 2025-06-29 by github-actions[bot]