FreeCAD/FreeCAD #12917: Flickering windows when mouse over

the16bitgamer, 2024-03-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Might be related to issue #12622 as I am using a nvidia GPU. However that was running on Linux whereas this is on Windows.

While mousing over intractable elements like the navigation cube, model elements, faces, or the file tree, the Window will flicker black for a single frame. This occurs whether the application is rendering in Software or hardware. Here is sample footage of the issue in action. https://youtu.be/vj3BU9dZQyI

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
[/code]
```

### Subproject(s) affected?

PartDesign

### Anything else?

I've tried to replicate this issue on Linux on another machine however this doesn't appear there.

My hardware which is impacted is a AMD Ryzen 5 1600x, 16GB RAM, Nvidia RTX 3050 with the software running on an SSD. I've updated both the OS and Graphics Drivers with no changes.

The other hardware I tested was a Intel® Core™ i5-1155G7, 16GB RAM, with the software running on an SSD in Linux Mint.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-13:
Could you please try this with the current weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

xtemp09, 2024-03-13:
@the16bitgamer, could you try _Mesa_ for Windows? I've never used it on Windows. =/

the16bitgamer, 2024-03-13:
> Could you please try this with the current weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

@maxwxyz I downloaded the latest weekly build from the link provided here is the about which freecad generates

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/Canada (en_CA)

Unfortunately the results are still the same. Occasional Flicker while hovering over a new UI element in Hardware OpenGL, which is worsened when switching over to Software OpenGL. Here is the video of me testing it: https://youtu.be/tREg4SplhYo

Everspace, 2024-05-01:
I was similarly impacted on my windows machine until I updated my nvidia drivers to 552.22 @the16bitgamer 

I have a NVIDIA GeForce RTX 3070 Ti

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * Alternate_OpenSCAD 1.0.0
```

I'm not super sure it's gone but it seems far better in the quick test I've done so far.

the16bitgamer, 2024-05-01:
@Everspace, just booted back into my windows install and can confirm that the latest update from Nvidia Driver 552.22 fixes this issue. I too am apprehensive that this is fixed, so I will leave it to the FreeCAD Team to either keep the case open or the close it.

On Linux Mint Nvidia driver 535 & 545 works without issue.

maxwxyz, 2024-05-06:
so could this issue be closed as driver issue?

the16bitgamer, 2024-05-07:
@maxwxyz Probably? It doesn't feel like a driver issue to me, since this was the only program on my machine to glitch out like this. But as I have no way to check (and I've moved away from Windows), I would consider this issue to be resolved.

I'll close this issue, but is there a way to put a note on it? Just in case someone else still has this issue, and it turns out that it's more than a Nvidia driver problem.

closed 2024-05-07 by the16bitgamer