FreeCAD/FreeCAD #12925: Sketcher Segmentation fault, GeoId index out range

ajohn77, 2024-03-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Freecad crashes when editing sketch in specific project containing just that simple sketch
1. Open attached project in "balkon_drzak.zip"
2. Open sketch
3. Zoom
4. Program crashes

Sketch has been originally normally created and after few **constrains and elements were removed** sketcher has started announcing various GeoId related errors. Sketcher stopped working properly then.
After restart that sketcher always causes Segmentation fault every time that sketch is edited (e.g. zoomed)

back trace:

```
[New Thread 0x7fff3b9fc640 (LWP 11948)]
<Exception> Sketch::checkGeoId. GeoId index out range.

Thread 1 "freecad" received signal SIGSEGV, Segmentation fault.
0x00007fff406587f1 in SketcherGui::EditModeGeometryCoinManager::updateGeometryColor(Sketcher::GeoListModel<std::unique_ptr<Sketcher::GeometryFacade const, std::default_delete<Sketcher::GeometryFacade const> > > const&, bool) () from /usr/lib/freecad-python3/lib/SketcherGui.so
```

appropriate project file which causes crashes:
[balkon_drzak.zip](https://github.com/FreeCAD/FreeCAD/files/14585671/balkon_drzak.zip)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-03-13:
0.22dev build crashes too.

<details><summary>Backtrace</summary>

```ruby
<Exception> Sketch::checkGeoId. GeoId index out range.
Process 1687 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
    frame #0: 0x00007fff592cf806 SketcherGui.so`SketcherGui::EditModeGeometryCoinManager::updateGeometryColor(Sketcher::GeoListModel<std::unique_ptr<Sketcher::GeometryFacade const, std::default_delete<Sketcher::GeometryFacade const>>> const&, bool) [inlined] std::vector<int, std::allocator<int>>::operator[](__n=0, this=size=0) at stl_vector.h:1126:39
   1123       operator[](size_type __n) _GLIBCXX_NOEXCEPT
   1124       {
   1125         __glibcxx_requires_subscript(__n);
-> 1126         return *(this->_M_impl._M_start + __n);
   1127       }
   1128
   1129       /**
```

```cpp
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
  * frame #0: 0x00007fff592cf806 SketcherGui.so`SketcherGui::EditModeGeometryCoinManager::updateGeometryColor(Sketcher::GeoListModel<std::unique_ptr<Sketcher::GeometryFacade const, std::default_delete<Sketcher::GeometryFacade const>>> const&, bool) [inlined] std::vector<int, std::allocator<int>>::operator[](__n=0, this=size=0) at stl_vector.h:1126:39
    frame #1: 0x00007fff592cf7fe SketcherGui.so`SketcherGui::EditModeGeometryCoinManager::updateGeometryColor(Sketcher::GeoListModel<std::unique_ptr<Sketcher::GeometryFacade const, std::default_delete<Sketcher::GeometryFacade const>>> const&, bool) [inlined] SketcherGui::CoinMapping::getPointGeoId(layerindex=0, pointindex=0, this=0x0000555557411a50) at EditModeCoinManagerParameters.h:490:53
    frame #2: 0x00007fff592cf7e8 SketcherGui.so`SketcherGui::EditModeGeometryCoinManager::updateGeometryColor(this=0x0000555558d3d980, geolistfacade=0x00007fffffffac30, issketchinvalid=false) at EditModeGeometryCoinManager.cpp:187:58
    frame #3: 0x00007fff592caeba SketcherGui.so`SketcherGui::EditModeCoinManager::updateColor(this=0x00005555574117c0, geolistfacade=0x00007fffffffac30) at EditModeCoinManager.cpp:892:54
    frame #4: 0x00007fff592caf7d SketcherGui.so`SketcherGui::EditModeCoinManager::updateColor(this=0x00005555574117c0) at EditModeCoinManager.cpp:885:16
    frame #5: 0x00007fff592e5807 SketcherGui.so`SketcherGui::ViewProviderSketch::updateColor(this=0x00005555594184a0) at ViewProviderSketch.cpp:2571:33
    frame #6: 0x00007fff592e9eac SketcherGui.so`SketcherGui::ViewProviderSketch::onSelectionChanged(this=0x00005555594184a0, msg=0x00007fffffffb340) at ViewProviderSketch.cpp:2190:30
    frame #7: 0x00007ffff7189ea3 libFreeCADGui.so`Gui::SelectionObserver::_onSelectionChanged(this=<unavailable>, msg=<unavailable>) at Selection.cpp:152:27
    frame #8: 0x00007ffff719864f libFreeCADGui.so`void std::__invoke_impl<void, void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&>((null)=__invoke_memfun_deref @ 0x00007fffffffafa0, __f=<unavailable>, __t=<unavailable>, (null)=<unavailable>) at invoke.h:74:46
    frame #9: 0x00007ffff719866d libFreeCADGui.so`boost::detail::function::void_function_obj_invoker1<std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>, void, Gui::SelectionChanges const&>::invoke(boost::detail::function::function_buffer&, Gui::SelectionChanges const&) [inlined] std::__invoke_result<void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&>::type std::__invoke<void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&>((null)=0x00007fffffffb340, (null)=<unavailable>, __fn=<unavailable>) at invoke.h:96:40
    frame #10: 0x00007ffff7198668 libFreeCADGui.so`boost::detail::function::void_function_obj_invoker1<std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>, void, Gui::SelectionChanges const&>::invoke(boost::detail::function::function_buffer&, Gui::SelectionChanges const&) [inlined] void std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>::__call<void, Gui::SelectionChanges const&, 0ul, 1ul>((null)=<unavailable>, __args=0x00007fffffffafa8, this=<unavailable>) at functional:506:24
    frame #11: 0x00007ffff7198664 libFreeCADGui.so`boost::detail::function::void_function_obj_invoker1<std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>, void, Gui::SelectionChanges const&>::invoke(boost::detail::function::function_buffer&, Gui::SelectionChanges const&) [inlined] void std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>::operator()<Gui::SelectionChanges const&, void>((null)=0x00007fffffffb340, this=<unavailable>) at functional:591:32
    frame #12: 0x00007ffff7198660 libFreeCADGui.so`boost::detail::function::void_function_obj_invoker1<std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>, void, Gui::SelectionChanges const&>::invoke(function_obj_ptr=<unavailable>, a0=0x00007fffffffb340) at function_template.hpp:158:11
[... cut ...]
    frame #95: 0x00007ffff3229e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:105:1), argc=1, argv=0x00007fffffffdd98, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdd88) at libc-start.c:392:3
    frame #96: 0x0000555555569c05 FreeCAD`_start + 37
(lldb)
```

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36338 (Git)
Build type: Debug
Branch: master
Hash: 5b5be480a3a14963cbdefe2e7a3923f9136af6a9
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

The backtrace is close to #12300 and #9198.

xtemp09, 2024-03-13:
This crash occurs because a function accesses an empty `std::vector`; abdullahtahiriyo wrote this.

Before accessing an `std::vector`, I would check first if it contains something. Perhaps, there was a preliminary checking that disappeared during refactoring. I hope masters of _Sketcher_ will fix this in no time.

FlachyJoe, 2024-03-13:
I'll take a look this (french) evening.

FlachyJoe, 2024-03-13:
The problem is due to a coincident constraint with only one point defined, I don't know how it can appear.

In document.xml:
`                        <Constrain Name="" Type="1" Value="0.0000000000000000" First="19" FirstPos="2" Second="-2000" SecondPos="0" Third="-2000" ThirdPos="0" LabelDistance="10.0000000000000000" LabelPosition="0.0000000000000000" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />`

The vector check as
```cpp
    int getPointGeoId(int pointindex, int layerindex)
    {
        if (PointIdToGeoId.empty()
            || layerindex > (signed)PointIdToGeoId.size()+1
            || pointindex > (signed)PointIdToGeoId[layerindex].size()+1){
            throw Base::IndexError("getPointGeoId. index out range.");
        }
        return PointIdToGeoId[layerindex][pointindex];
    }
```

solve the crash but then the sketch can't leave the edit mode.

I think we have to validate the constraint at document opening.

luzpaz, 2024-09-03:
@wwmayer if you have a moment, could you weigh-in ?

luzpaz, 2024-09-30:
softest of bumps

PaddleStroke, 2025-05-26:
FYI it's fixed in wmayer's branch : https://github.com/FreeCAD/FreeCAD/commit/b96a786625cf2013f664c13482cbdc670d551218