FreeCAD/FreeCAD #12934: Some input fields in FEM MaterialFluid and FEM MaterialSolid are getting reset automatically

FEA-eng, 2024-03-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the current weekly builds, some input fields of [FEM MaterialFluid](https://wiki.freecad.org/FEM_MaterialFluid) and [FEM MaterialSolid](https://wiki.freecad.org/FEM_MaterialSolid) (available after checking the "use this task panel" box) are not working properly - they are getting reset when the default value is changed. The following input fields are affected:
- FEM MaterialFluid: Kinematic viscosity (this one is the worst because it's reset to 0, preventing fluid analyses from running), Specific Heat
- FEM MaterialSolid: Specific Heat (but it seems to be inconsistent - it doesn't always happen)

@dave

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davesrocketshop, 2024-03-13:
On the surface this is unlikely a materials issue, but I'll look into it to verify

FEA-eng, 2024-03-13:
Update: it seems that the Specific heat resets also in 0.21.2 but there are no such issues with Kinematic viscosity in the stable release.

FEA-eng, 2024-06-15:
@marioalexis84 It seems to be more related to FEM itself than to Materials. Can you have a look at this at some point?

luzpaz, 2024-07-29:
@marioalexis84 soft bump

FEA-eng, 2024-09-03:
@bgbsww As we discussed on Discord, I'm pinging you on this issue with steps to reproduce:

1. Go to FEM workbench.
2. Add [Analysis container](https://wiki.freecad.org/FEM_Analysis).
3. Add [Material for fluid](https://wiki.freecad.org/FEM_MaterialFluid).
4. Check the "Use this task panel" box.
5. Type 1200 for Kinematic Viscosity. Click away from this field. It changes unit and shows zero due to the default decimal precision. Click OK. Go back to MaterialFluid. Now it shows 1 mm^2/s again for some reason so it's reset.
6. Now just change the default 2050 for Specific Heat Capacity to e.g. 5050. Click away from it. It will reset to the previous value.

It seems to happen also with Thermal Conductivity and with the same quantities in [Material for solid](https://wiki.freecad.org/FEM_MaterialSolid)

bgbsww, 2024-09-03:
Interesting.  So far I cannot duplicate on my local debug build of main. which won't even allow 1200 - it caps at 999.  The field never changes on me.   Tried again on a daily and got the same thing.

Stepping through update_material_property in task_material_common.py shows an inputfield_text of '120.00 mm^2/s' as I would expect.

Specific heat is showing as 4182000000.00 mm^2/(s^2*K) but when I copied it to paste here I got: '2050000000.00 mm^2/(s^2*K)'  which would seem like a clue.  And when I focused back it had that value, which is one order of magnitude too large to input - try to edit and it fails.   505000000.00 will be accepted, and again doesn't change.

Looking at my general preferences units are metric small parts,  so switched to Standard...

Now 120 is in m^2/s and throws a Unit mismatch in update_material_properties at line 562 for me: `            value = Units.Quantity(inputfield_text).getValueAs(qUnit)`  closes the dialog, and when reopened is reset to 1 mm^2

So I think this might be dependent on which unit preferences you have set, and then how the fields behave and parse.

OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38501 (Git)
Build type: Debug
Branch: bgbsww-toponamingRevolution
Hash: 4ea2a0162d2acf74679864d2ce865e7fdd672f1b
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.4.24
  * Curves 0.6.45
  * DynamicData 2.62.0
  * BIM021 2021.12.0
  * parts_library

marioalexis84, 2024-09-03:
Thanks @FEA-eng . I hadn't been able to reproduce it until now.

FEA-eng, 2024-09-03:
> So I think this might be dependent on which unit preferences you have set, and then how the fields behave and parse.

Thanks for checking. We do have issues with some units in non-default unit systems but the default one (with millimeters) should work fine in all cases. This issue seems to be a nasty exception.

PR #16380 (2024-09-08): FEM: Fluid material properties fix
fully fixes #12934

This fix consists of 3 parts:
- check of empty string fix from @benj5378 (#16270)
- higher maximum values - they were way too low which was a nasty bug/oversight: https://github.com/FreeCAD/FreeCAD/pull/16270#issuecomment-2329560196
- correct writing of viscosity to CalculiX input files (it was broken and did not use kinematic viscosity)

@marioalexis84 FYI
closed 2024-09-08 by chennes