FreeCAD/FreeCAD #12935: Some FEM examples show errors when trying to open the material definition

FEA-eng, 2024-03-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the recent weekly builds, double-clicking on a material object in FEM examples may result in errors.

Steps to reproduce:
1. Go to the FEM workbench, Utilities --> Open FEM examples
2. Select Solvers --> calculix --> Box Analysis Static and click Setup.
3. The analysis can be submitted but if you double-click on the material definition, you will see:

```
19:42:51  pyException: Traceback (most recent call last):
  File "E:\FreeCAD_weekly-builds-36391-2024-03-11-conda-Windows-x86_64-py310\Mod\Fem\femviewprovider\view_material_common.py", line 58, in setEdit
    view_base_femconstraint.VPBaseFemConstraint.setEdit(
  File "E:\FreeCAD_weekly-builds-36391-2024-03-11-conda-Windows-x86_64-py310\Mod\Fem\femviewprovider\view_base_femobject.py", line 93, in setEdit
    task = TaskPanel(vobj.Object)
  File "E:\FreeCAD_weekly-builds-36391-2024-03-11-conda-Windows-x86_64-py310\Mod\Fem\femtaskpanels\task_material_common.py", line 160, in __init__
    self.card_path = self.get_material_card(self.material)
  File "E:\FreeCAD_weekly-builds-36391-2024-03-11-conda-Windows-x86_64-py310\Mod\Fem\femtaskpanels\task_material_common.py", line 290, in get_material_card
    if float(item[1].split()[0]) == float(a_mat_item[1].split()[0]):
<class 'ValueError'>: could not convert string to float: '2700,00'
```

The same happens e.g. with Solvers --> elmer --> Electrostatic Capacitance Two Balls and with several other examples but not with Solvers --> calculix --> Lateral Torsional Buckling, for instance.

I don't know why some examples work and some don't.

@davesrocketshop

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davesrocketshop, 2024-03-13:
The problem is that the Python `float` conversion is not locale aware. See https://peps.python.org/pep-0331/ for an explanation.

The function you want to use is `locale.atof()`

This likely occurs in other parts of the code as well.

davesrocketshop, 2024-03-13:
Just to clarify, having looked into v0.21 as well, the value in question '2700,00' is not contained in the Steel-Generic.FCMat material card. It comes from somewhere within the FEM code.

davesrocketshop, 2024-03-13:
digging even deeper, it is comparing the selected card with others, so it may occur in another material card. I need to investigate more. The dictionary creation may be doing an unintended locale based conversion.

davesrocketshop, 2024-03-13:
Definitely in the dictionary creation. The material `getString()` function returns a localized version for Quantity types rather than the original string. This should be a quick fix.

PR #12937 (2024-03-15): Material: Removes localization from the material card dictionary
Fixes #12935

The previous card implementation transferred strings directly to the card dictionary without interpretation. The new material system parses the file data, producing a dictionary for compatibility. The new dictionaries localized the strings which is incorrect. This PR removes that localization.
commit 48bfd227 (2024-03-15): Material: Material editor enhancements

Removes localization from the material card dictionary.

Fixes #12935

The previous card implementation transferred strings directly to the
card dictionary without interpretation. The new material system parses
the file data, producing a dictionary for compatibility. The new
dictionaries localized the strings which is incorrect. This PR
removes that localization.
closed 2024-03-15 by chennes