FreeCAD/FreeCAD #12936: New Standard Part & Assembly are considered modified documents

pierreporte, 2024-03-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

How to reproduce:

1. Open FreeCAD
2. Click on Standard Part or Assembly
3. Close the new document
4. You are asked if you want to save the file or not

Users shouldn’t be asked if they want to save because they have done absolutely nothing with the new document.

2D Draft and Architecture do not have the problem because they don’t put anything in the tree.

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36331 (Git) AppImage
Build type: Release
Branch: main
Hash: eb5a1b333946ce4ac8a4ead835cba8bdcf602a87
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * CosmeticThread3D
  * TabBar (Disabled)
  * Curves 0.6.23
  * CurvedShapes 1.0.5
  * sheetmetal 0.4.3
  * OpenDark 2023.12.1
  * OpenLight 2023.12.17
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-03-13:
> Click on Standard Part or Assembly

Adding a Part or Assembly is not modifying the document?

pierreporte, 2024-03-13:
From an user point of view, they aren’t added in this situation. Opening a template should feel like creating an empty document. Of course, starting from an empty file and adding a part or an assembly is definitely modifying it.