FreeCAD/FreeCAD #12944: [Feature Request] Allow adding text to the sketch

uzytkownik, 2024-03-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently the only way of adding text to FreeCAD is StringShape which doesn't play nicely with parametric design (e.g. font cannot be specified as function, neither can size as far as I can tell) and it's painful to use with Part Design.

It would be good if user could just type string inside sketcher and extrude it if they use Part Design workflow.

### Full version info

```shell
[code]
OS: NixOS 24.05 (Uakari) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.2.Unknown
Build type: Release
Python 3.11.8, Qt 5.15.12, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.22.9
  * Assembly4 0.50.6
  * fasteners 0.5.0
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-04-08:
Right, this would be very useful. From SW:

[image]

dvdmrtnz, 2024-08-20:
This is the one feature I miss most from SolidWorks. It would be great to have it!

pierreporte, 2024-08-26:
This feature can be useful. It would be good if the implementation could understand open-type features, at least kerning.

luzpaz, 2024-08-26:
It would be a shame to have to 're-invent the wheel' here. Inkscape does this very well.

caio-venancio, 2025-03-06:
https://forum.freecad.org/viewtopic.php?p=815130

Hey,
I'm starting a topic for GSoC 2025 to explore the potential of this feature. If anyone here could contribute to the discussion, I’d really appreciate it!
Tagging: @uzytkownik @pierreporte @luzpaz @maxwxyz

dubstar-04, 2025-06-27:
This is awesome...