FreeCAD/FreeCAD #12946: Segfault when deleting dimension if dimension formatting dialogue is still open

mpfusion, 2024-03-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When deleting a dimension in TechDraw when the dimension formatting/display dialogue is open FreeCAD segfaults when the dialogue is closed.

To reproduce:
- create a horizontal/vertical dimension in TechDraw or open attached drawing
- double click on the dimension, which opens the formatting/display dialogue
- hit <delete> when the dimension is still selected
→ the dimension is gone, but the dialogue window is still open
- close the dimension dialogue with “Cancel”
→ segfault 
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c050) [0x7f49d32c5050]
#1  0x7f496b2f48c8 in TechDrawGui::TaskDlgDimension::reject() from /home/mp/usr/bin/freecad-bin/lib/TechDrawGui.so+0x48
#2  0x7f49d6443a46 in Gui::TaskView::TaskView::reject() from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x56
#3  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2e8f4f) [0x7f49d398cf4f]
#4  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2fe860) [0x7f49d45c5860]
#5  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2e8f7c) [0x7f49d398cf7c]
#6  0x7f49d451bfc2 in QAbstractButton::clicked(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x42
#7  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x25522a) [0x7f49d451c22a]
#8  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x256db8) [0x7f49d451ddb8]
#9  0x7f49d451dfd7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf7
#10  0x7f49d446bdb8 in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1a8
#11  0x7f49d4429fae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#12  0x7f49d4432552 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x13c2
#13  0x7f49d60920b8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x68
#14  0x7f49d39556f8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#15  0x7f49d443065e in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1be
#16  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1bdbd8) [0x7f49d4484bd8]
#17  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c0f60) [0x7f49d4487f60]
#18  0x7f49d4429fae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#19  0x7f49d60920b8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x68
#20  0x7f49d39556f8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#21  0x7f49d3d343ed in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x80d
#22  0x7f49d3d08cac in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#23  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x6deca) [0x7f49cd596eca]
#24  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x299) [0x7f49d16f27a9]
#25  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x54a38) [0x7f49d16f2a38]
#26  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f49d16f2acc]
#27  0x7f49d39ad836 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x66
#28  0x7f49d395417b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#29  0x7f49d395c2d6 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x96
#30  0x7f49d5fda496 in Gui::Application::runApplication() from /home/mp/usr/bin/freecad-bin/lib/libFreeCADGui.so+0x756
#31  freecad-dev(+0x8f0e) [0x5566155f7f0e]
#32  /lib/x86_64-linux-gnu/libc.so.6(+0x2724a) [0x7f49d32b024a]
#33  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f49d32b0305]
```

https://forum.freecad.org/viewtopic.php?t=86109

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36404 (Git)
Build type: Release
Branch: main
Hash: 69a3371e617ec44b9af45f34622fbf4180c91c24
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Ireland (en_IE)
Installed mods: 
  * BOLTSFC
  * sheetmetal 0.4.3
  * Assembly4 0.50.9
  * kicadStepUpMod 11.1.0
```

### Subproject(s) affected?

None

### Anything else?

[delete-dimension-segfault.FCStd.tar.gz](https://github.com/FreeCAD/FreeCAD/files/14599520/delete-dimension-segfault.FCStd.tar.gz)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-14:
Can confirm on
<details><summary>Windows 11 0.22dev</summary>
<p>

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.24
  * Defeaturing 1.2.2
  * fasteners 0.5.14
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.3
```

</p>
</details> 

[image]

wwmayer, 2024-03-15:
This commit https://github.com/FreeCAD/FreeCAD/commit/29cfe6db51 is part of PR #12966 fixes the crash for the described workflow.

But with a trick this can be bypassed and the crash still occurs: https://forum.freecad.org/viewtopic.php?p=747232#p747232

WandererFan, 2024-03-18:
weak pointer implemented here: PR #13014

wwmayer, 2024-03-18:
I can confirm that it's fixed now.

closed 2024-03-18 by wwmayer