FreeCAD/FreeCAD #12950: Choosing a user-defined material in FEM_MaterialSolid makes it impossible to access the material object anymore

FEA-eng, 2024-03-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
1. In the Material Editor, select any built-in material with mechanical properties and click Inherit New to create a user-defined material with copied settings. Name it and save.
2. In the FEM_MaterialSolid tool, select this newly created material and click OK.
3. Double-clicking on the MaterialSolid object results in this error:

```
12:38:45  pyException: Traceback (most recent call last):
  File "E:\Program Files\FreeCAD_weekly-builds-36391-2024-03-11-conda-Windows-x86_64-py310\Mod\Fem\femviewprovider\view_material_common.py", line 58, in setEdit
    view_base_femconstraint.VPBaseFemConstraint.setEdit(
  File "E:\Program Files\FreeCAD_weekly-builds-36391-2024-03-11-conda-Windows-x86_64-py310\Mod\Fem\femviewprovider\view_base_femobject.py", line 93, in setEdit
    task = TaskPanel(vobj.Object)
  File "E:\Program Files\FreeCAD_weekly-builds-36391-2024-03-11-conda-Windows-x86_64-py310\Mod\Fem\femtaskpanels\task_material_common.py", line 160, in __init__
    self.card_path = self.get_material_card(self.material)
  File "E:\Program Files\FreeCAD_weekly-builds-36391-2024-03-11-conda-Windows-x86_64-py310\Mod\Fem\femtaskpanels\task_material_common.py", line 288, in get_material_card
    if not self.isfloat(item[1].split()[0]):
<class 'IndexError'>: list index out of range
```

@daves

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-06-13:
@davesrocketshop This issue is still present in the newest weekly. Can you have a look at it? It's quite nasty as custom materials can't be properly used in FEM because of it.

davesrocketshop, 2024-06-14:
> @davesrocketshop This issue is still present in the newest weekly. Can you have a look at it? It's quite nasty as custom materials can't be properly used in FEM because of it.

I'll look into it.

davesrocketshop, 2024-06-15:
The problem is in src/Mod/Fem/femtaskpanels/task_material_common.py line 288:
if not self.isfloat(item[1].split()[0]):

It appears it does not handle the case of an empty string (i.e. item[1] has no value)

FEA-eng, 2024-06-15:
> The problem is in src/Mod/Fem/femtaskpanels/task_material_common.py line 288: if not self.isfloat(item[1].split()[0]):
> 
> It appears it does not handle the case of an empty string (i.e. item[1] has no value)

@marioalexis84 Sorry for bothering you again but can you have a look at this too as it needs to be fixed from the FEM side?

PR #15179 (2024-07-15): FEM: Fix error with custom material card
Fixes #12950

@marioalexis84 and @davesrocketshop FYI
closed 2024-07-15 by chennes