FreeCAD/FreeCAD #12955: .scad import failure - cannot unpack non-iterable bool object

ScottCLo, 2024-03-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm trying to open my OpenSCAD project in FreeCAD and it fails with an error message about `<class 'TypeError'>: cannot unpack non-iterable bool object`

I've tested multiple files and tried it on two other Linux machines with the same result.

```
Msg: opening C:\Users\ROBSIG~1\AppData\Local\Temp\fc-04856-364999-000004.dxf...
Msg: There has been an error:Msg: 
Msg: First section is not HEADERMsg: 
Err: Traceback (most recent call last):
  File "C:\Users\Rob Sight\AppData\Local\Programs\FreeCAD 0.21\Mod\OpenSCAD\OpenSCADFeatures.py", line 592, in execute
    shape = OpenSCAD.OpenSCADUtils.process_ObjectsViaOpenSCADShape(fp.Document,fp.Children,\
  File "C:\Users\Rob Sight\AppData\Local\Programs\FreeCAD 0.21\Mod\OpenSCAD\OpenSCADUtils.py", line 648, in process_ObjectsViaOpenSCADShape
    return process2D_ObjectsViaOpenSCADShape(children,name,doc)
  File "C:\Users\Rob Sight\AppData\Local\Programs\FreeCAD 0.21\Mod\OpenSCAD\OpenSCADUtils.py", line 590, in process2D_ObjectsViaOpenSCADShape
    face = importDXFface(tmpfilename,None,None)
  File "C:\Users\Rob Sight\AppData\Local\Programs\FreeCAD 0.21\Mod\OpenSCAD\OpenSCAD2Dgeom.py", line 486, in importDXFface
    layers = importDXF.processdxf(doc,filename,False,False) or importDXF.layers
  File "C:\Users\Rob Sight\AppData\Local\Programs\FreeCAD 0.21\Mod\Draft\importDXF.py", line 2147, in processdxf
    drawing = dxfReader.readDXF(filename)
  File "C:\Users\Rob Sight\AppData\Roaming\FreeCAD\Macro\dxfReader.py", line 399, in readDXF
    (infile, drawing) = sm.run((infile, None))
<class 'TypeError'>: cannot unpack non-iterable bool object
```

[FreeCAD.log](https://github.com/FreeCAD/FreeCAD/files/14605286/FreeCAD.log)
[project.scad.txt](https://github.com/FreeCAD/FreeCAD/files/14605297/project.scad.txt)

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-14:
Could you also test this with the latest dev build: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Syres916, 2024-03-14:
The failure is being generated by _dxfReader.py_ which is not part of FreeCAD core, it's a set of modules downloaded from https://github.com/yorikvanhavre/Draft-dxf-importer Just from the line number of the failure the OP is not running the latest versions of Yorik's code, it would be line 403 but that's not a guarantee of it being fixed.

ScottCLo, 2024-03-15:
> Could you also test this with the latest dev build: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

I couldn't get it to run on windows or my Wayland linux system, i think it said there was a problem with Qt. It runs on my system running xorg though but has the same error, see log at bottom.

> The failure is being generated by _dxfReader.py_ which is not part of FreeCAD core, it's a set of modules downloaded from https://github.com/yorikvanhavre/Draft-dxf-importer Just from the line [...]

I cloned the repo and overwrote the stuff auto downloaded. Version 21.2 wont recognize it at all and asks if i want to auto download, witch just replaces it with the old version. If i run it on the Dev version it recognizes the up to date code but still fails to open the test file.

### Full dev version info
```
OS: Void Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git) AppImage
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```
[FreeCAD.log](https://github.com/FreeCAD/FreeCAD/files/14609985/FreeCAD.log)

maxwxyz, 2024-03-16:
maybe @yorikvanhavre could weigh in regarding the DXF reader?

bkw777, 2024-08-19:
I can't even import a simple hull with nothing but a sphere and cylinder.

```
$fa = $preview?3:1;
$fs = $preview?0.2:0.1;
hull() {
  sphere(d=4);
  cylinder(h=6,d=4);
}
```

```
23:34:04  WARNING: Token 'DOT' defined, but not used
23:34:04  WARNING: Token 'WORD' defined, but not used
23:34:04  WARNING: There are 2 unused tokens
23:34:04  End processing CSG file
23:34:04  hull: No shells or compsolids found in shape
```

tree view shows a hull with included sphere and cylinder, but greyed out and red exclaim, no model in the main display.

I can un-hide the individual objects but not the hull.

right-click recompute on the hull gives:
```
23:40:49  pyException: Traceback (most recent call last):
  File "/usr/share/freecad-daily/Mod/OpenSCAD/OpenSCADFeatures.py", line 554, in execute
    raise ValueError
<class 'ValueError'>: 
23:40:49  hull: 

```

```
OS: Ubuntu 23.10 (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38051 (Git)
Build type: Release
Branch: main
Hash: dfe0c5d4ee4841f11e62da1104e322e95834aba2
Python 3.11.6, Qt 5.15.10, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.5.18
  * BOLTSFC
  * Manipulator 1.5.7
  * fcgear 1.0.0
  * kicadStepUpMod 11.1.0
```

luzpaz, 2024-08-19:
CC @KeithSloan (any chance you could weigh-in ?)

KeithSloan, 2024-08-19:
Works on my system

[image]
[test2dHull.scad.txt](https://github.com/user-attachments/files/16662835/test2dHull.scad.txt)
[test2dhull.FCStd.txt](https://github.com/user-attachments/files/16662836/test2dhull.FCStd.txt)

As a 2d object is involved the importer will create a DXF file and there maybe a problem with the DXF import.
For the Alternate Importer Matt Considine changed to using the EzDXF library which is a big improvement over the
various FC importers.

But as I say works in my system even using the standard FC implementation of the importer

OS: macOS 14.4
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * ImportExport_3DM 0.1.0
  * DynamicData 2.62.0
  * Pyramids-and-Polyhedrons
  * occi-freecad-plugin 0.1.0
  * GDML 2.0.0
  * Design456 0.0.1
  * Alternate_OpenSCAD 1.0.0
  * SWPart
  * toSketch 1.0.1
  * Curves 0.6.42

KeithSloan, 2024-08-19:
> I can't even import a simple hull with nothing but a sphere and cylinder.
> 
> ```
> $fa = $preview?3:1;
> $fs = $preview?0.2:0.1;
> hull() {
>   sphere(d=4);
>   cylinder(h=6,d=4);
> }
> ```

Again works here 
[image]
[hulltest4.FCStd.txt](https://github.com/user-attachments/files/16663018/hulltest4.FCStd.txt)

Does fail with alternate importer but that is trying to be clever with its hull handling and I will look at that.

luzpaz, 2024-08-24:
@KeithSloan :point_up_2: :pray:

luzpaz, 2024-08-30:
> But as I say works in my system even using the standard FC implementation of the importer

Closing. If anyone still can reproduce on latest stable, please re-open.

DrTomFlint, 2025-03-23:
Adding a hull with 2 spheres fails using FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage running on Ubuntu 22.04.

uname -a shows Linux LevLab6 6.8.0-52-generic #53~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Jan 15 19:18:46 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

0) start a fresh copy of freecad
1) create a new file
2) switch to openscad workbench
3) add OpenSCAD Element
4) type into the code box:

hull(){
  sphere(r=10);
  translate([0,0,40])
  sphere(r=5);
}

5) hit the Add button.
6) report view shows "hull: No shells or compsolids found in shape"

Nothing is show in the 3D view, nothing is added to the Model tab.
Screenshot is attached.

[image]

DrTomFlint, 2025-03-23:
Same steps work fine in FreeCAD 0.19

KeithSloan, 2025-03-23:
Works fine  for me on 1.0.0

OS: macOS 15.3.2
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * ImportNURBS 1.1.0
  * ImportExport_3DM 0.1.0
  * CAD_Files_Git
  * RemoteServerManager 1.0.1
  * OpenTheme 2024.9.1
  * CfdOF 1.29.8
  * GDML 2.0.1Beta
  * SVG_FC_Object
  * Alternate_OpenSCAD 1.0.0
  * toSketch 1.0.1

[testHull2.zip](https://github.com/user-attachments/files/19411117/testHull2.zip)

also original two circles

[TestHull.zip](https://github.com/user-attachments/files/19411135/TestHull.zip)

bkw777, 2025-03-24:
Failed for me too but this setting made it work:  

Switch to OpenSCAD workbench.  
Edit -> Preferences  -> OpenSCAD  
  Send to OpenSCAD via: "Standard temp directory"

Now it works.

However something is still wrong. The pipe option is still broken even if you have openscad >= 2021.1

At first I thought it was because the ubuntu 24.04 repos provide openscad "2021.01"

fails:  
OpenSCAD executable: "/usr/bin/openscad"  
Send to OpenSCAD via: "stdout pipe (requires OpenSCAD >= 2021.1)"  
```
$ /usr/bin/openscad --version
OpenSCAD version 2021.01
```

But I also tried with openscad 2025.03.23.nightly and it still fails with stdout.  

still fails:  
OpenSCAD executable: "/usr/bin/openscad-nightly"  
Send to OpenSCAD via: "stdout pipe (requires OpenSCAD >= 2021.1)"  
```
$ /usr/bin/openscad-nightly --version
OpenSCAD version 2025.03.23.nightly
```

So, regardless what version of openscad you have:  

Switch to OpenSCAD workbench.  
Edit -> Preferences  -> OpenSCAD  
Send to OpenSCAD via: "Standard temp directory"

Works.

However, curiously, the sphere and translate, with only the hull commented out works fine on both versions of openscad.

So only hull doesn't work, and only with stdout pipe.
Pipe seems to work fine for other openscad code, just not hull() ???

```
OS: Ubuntu 24.04.2 LTS (XFCE/xubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40913 (Git)
Build date: 2025/03/22 17:27:24
Build type: Release
Branch: main
Hash: 3b6cb37dd9343cda0d87c56748b0f1a8f8eb9cfd
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/201.061/1
Installed mods: 
  * Manipulator 1.5.7
  * BOLTSFC
  * kicadStepUpMod 11.4.4
  * fasteners 0.5.33
  * fcgear 1.0.0
```

KeithSloan, 2025-03-24:
If you check Report View are you seeing any error messages?
[code]
22:20:35  Process OpenSCAD Shapes via OpenSCAD
22:20:35  Error all shapes must be either 2D or both must be 3D
22:20:35  pyException: Traceback (most recent call last):
[/code]

KeithSloan, 2025-03-27:
"So only hull doesn't work, and only with stdout pipe.
Pipe seems to work fine for other openscad code, just not hull() ???

Should this maybe be a new more specific issue just about the pipe option now?"

Maybe something to do with the "encoding ?" option used on the pipe request?

Apart from an initial invocation of the pipe to OpenSCAD to create a csg from a scad file and return it, the only other use is for hull, Minkowski operations, other functions parsed in csg are dealt with "locally" i.e. in the importer within FreeCAD

tristanbay, 2025-08-25:
Nope, still crashes. I get a slightly shorter backtrace but the segfault is still there.

I'm not comfortable sharing my full private OpenSCAD code but I'll share the module I wrote with the first instance of `hull()` in the file:

```module top_rounded_block(x1y1z,x2y2,r) {
	translate([-x1y1z[0]/2,-x1y1z[1]/2,0]) {
		hull() {
			translate([r,r,0]) cylinder(0.001,r,0);
			translate([x1y1z[0]-r,r,0]) cylinder(0.001,r,0);
			translate([r,x1y1z[1]-r,0]) cylinder(0.001,r,0);
			translate([x1y1z[0]-r,x1y1z[1]-r,0]) cylinder(0.001,r,0);
			
			translate([
				r+((x1y1z[0]-x2y2[0])/2),
				r+((x1y1z[1]-x2y2[1])/2),
				x1y1z[2]-r]
			) sphere(r);
			
			translate([
				x1y1z[0]-r-((x1y1z[0]-x2y2[0])/2),
				r+((x1y1z[1]-x2y2[1])/2),
				x1y1z[2]-r
			]) sphere(r);
			
			translate([
				r+((x1y1z[0]-x2y2[0])/2),
				x1y1z[1]-r-((x1y1z[1]-x2y2[1])/2),
				x1y1z[2]-r
			]) sphere(r);
			
			translate([
				x1y1z[0]-r-((x1y1z[0]-x2y2[0])/2),
				x1y1z[1]-r-((x1y1z[1]-x2y2[1])/2),
				x1y1z[2]-r
			]) sphere(r);
		}
	}
}
```

Merlin04, 2025-10-06:
I'm getting this error in FreeCAD 1.0.2 when trying to load [`x-end-motor.scad` from the Prusa i3 MK3S source](https://github.com/prusa3d/Original-Prusa-i3/blob/MK3S/Printed-Parts/SCAD/x-end-motor.scad):

```
21:32:57  WARNING: variable center not specified as parameter in file x-end-motor.scad, line 106

21:32:57  WARNING: Token 'DOT' defined, but not used
21:32:57  WARNING: Token 'WORD' defined, but not used
21:32:57  WARNING: There are 2 unused tokens
21:32:57  obj [<Part::PartFeature>] has no Name & Shape
21:32:57  obj [<Part::PartFeature>] has no Name & Shape
21:32:57  Recompute failed : difference
21:32:57  <Exception> Input shape is null
21:32:57  Recompute failed : Group001
21:32:57  <Exception> Input shape is null
21:32:57  Recompute failed : Group001
21:32:57  obj [<Part::PartFeature>] has no Name & Shape
21:32:57  obj [<Part::PartFeature>] has no Name & Shape
21:32:57  obj [<Part::PartFeature>] has no Name & Shape
21:32:57  obj [<Part::PartFeature>] has no Name & Shape
21:32:57  obj [<Part::PartFeature>] has no Name & Shape
21:32:57  obj [<Part::PartFeature>] has no Name & Shape
21:32:57  obj [<Part::PartFeature>] has no Name & Shape
21:32:57  Recompute failed : difference002
21:32:57  Recompute failed : difference005
21:32:57  <Exception> Input shape is null
21:32:57  Recompute failed : Group006
21:32:57  <Exception> Input shape is null
21:32:57  Recompute failed : Matrix_Union
21:32:57  <Exception> Input shape is null
21:32:57  Recompute failed : Group006
21:32:57  <Exception> Input shape is null
21:32:57  Recompute failed : Matrix_Union
21:32:57  textmsg : text ( text="R6", size = 4, spacing = "1", font = "helvetica:style=Bold", direction = "ltr", language = "en", script = "Latn", halign = "left", valign = "baseline", $fn = 0, $fa = 12, $fs = 2);
21:32:57  Geometries in cache: 1
Geometry cache size in bytes: 1616
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.015
   Top level object is a 2D object:
   Contours:        4

21:33:01  opening /tmp/fc-2044874-517690-000002.dxf...
21:33:01  There has been an error:
21:33:01  First section is not HEADER
21:33:01  pyException: Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "/tmp/.mount_FreeCAfnFjid/usr/lib/python3.11/site-packages/freecad/module_io.py", line 16, in OpenInsertObject
    getattr(importerModule, importMethod)(objectPath, *importArgs, **importKwargs)
  File "/tmp/.mount_FreeCAfnFjid/usr/Mod/OpenSCAD/importCSG.py", line 120, in open
    processcsg(tmpfile)
  File "/tmp/.mount_FreeCAfnFjid/usr/Mod/OpenSCAD/importCSG.py", line 178, in processcsg
    result = parser.parse(f.read())
             ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/me/.local/lib/python3.11/site-packages/ply/yacc.py", line 333, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/me/.local/lib/python3.11/site-packages/ply/yacc.py", line 1120, in parseopt_notrack
    p.callable(pslice)
  File "/tmp/.mount_FreeCAfnFjid/usr/Mod/OpenSCAD/importCSG.py", line 1313, in p_text_action
    p[0] = [processTextCmd(t)]
            ^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAfnFjid/usr/Mod/OpenSCAD/importCSG.py", line 954, in processTextCmd
    face = importDXFface(tmpfilename,None,None)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAfnFjid/usr/Mod/OpenSCAD/OpenSCAD2Dgeom.py", line 486, in importDXFface
    layers = importDXF.processdxf(doc,filename,False,False) or importDXF.layers
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAfnFjid/usr/Mod/Draft/importDXF.py", line 2227, in processdxf
    drawing = dxfReader.readDXF(filename)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/me/.local/share/FreeCAD/Mod/dxf-library/dxfReader.py", line 403, in readDXF
    (infile, drawing) = sm.run((infile, None))
    ^^^^^^^^^^^^^^^^^
<class 'TypeError'>: cannot unpack non-iterable bool object
```

I've tried this with "Use legacy Python importer" both checked and unchecked, but it still errors.

Knochi, 2025-11-25:
yesterday it worked but today i have issues again with 2D objects.
Same AppImage for FreeCAD and openSCAD.. no idea what changed.
I tried the flatpak FreeCAD install in between, maybe that messed up something.

seems that the DXF importer is not working as expected this time

> Opening /tmp/fc-11446-675007-000297.dxf...
> drawing 1 polylines...
> Traceback (most recent call last):
>   File "/tmp/.mount_FreeCAfGGHhL/usr/Mod/OpenSCAD/OpenSCADFeatures.py", line 550, in execute
>     shape = OpenSCAD.OpenSCADUtils.process_ObjectsViaOpenSCADShape(fp.Document,fp.Children,\
>             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/tmp/.mount_FreeCAfGGHhL/usr/Mod/OpenSCAD/OpenSCADUtils.py", line 644, in process_ObjectsViaOpenSCADShape
>     return process2D_ObjectsViaOpenSCADShape(children,name,doc)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/tmp/.mount_FreeCAfGGHhL/usr/Mod/OpenSCAD/OpenSCADUtils.py", line 586, in process2D_ObjectsViaOpenSCADShape
>     face = importDXFface(tmpfilename,None,None)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/tmp/.mount_FreeCAfGGHhL/usr/Mod/OpenSCAD/OpenSCAD2Dgeom.py", line 445, in importDXFface
>     layers = importDXF.processdxf(doc,filename,False,False) or importDXF.layers
>              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/tmp/.mount_FreeCAfGGHhL/usr/Mod/Draft/importDXF.py", line 2325, in processdxf
>     shape = drawPolyline(polyline, num=num)
>             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/tmp/.mount_FreeCAfGGHhL/usr/Mod/Draft/importDXF.py", line 969, in drawPolyline
>     elif (dxfCreateDraft or dxfCreateSketch) and (not curves) and (not forceShape):
>           ^^^^^^^^^^^^^^
> <class 'NameError'>: name 'dxfCreateDraft' is not defined
> hull002: name 'dxfCreateDraft' is not defined

Knochi, 2025-11-28:
> Thanks for reporting, going to be busy till after the weekend. As a workaround you can import as a SCAD File Object. (Best to also set an external editor - see README)

tried that as well with similar error messages.

KeithSloan, 2025-11-28:
You mean the the Housing.FCStd file does not work for you?

KeithSloan, 2025-11-28:
As aside looking at improving things with new External OpenSCAD workbench for FC 1.2 which plans to drop the internal OpenSCAD workbench. https://github.com/KeithSloan/OpenSCAD_Workbench

Used a ChatGPT session to discuss if functions in BSOL/BSOL2 could be replaced by BREP friendly ones.
i.e. Have same calls but avoid use of hull.

See session pdf and txt files https://github.com/KeithSloan/OpenSCAD_Workbench/tree/main/freecad/OpenSCAD_Ext/Developer_Notes.

Are you using BSOL or BSOL2?

If yes how easy would it be for you to the test txt(changed to SCAD files)?