FreeCAD/FreeCAD #12958: Assembly: doubleClicked to edit joint & cancel button. Transaction not aborted

PaddleStroke, 2024-03-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In assembly double clicking a joint start a taskbox to edit the joint.
Pressing cancel does not revert the transaction.
See the explanation in the following video:

https://github.com/FreeCAD/FreeCAD/assets/19984177/e1d9b398-9012-4c62-8bd6-467e8665bfc7

I think what happens is that the 'Label edition' transaction comes after I open a transaction in my task class. So it closes my transaction. Then Label edition abort itself.

By any luck do you happen to know what is happening @wwmayer please?

For reference : 

```
    def doubleClicked(self, vobj):
        task = Gui.Control.activeTaskDialog()
        if task:
            task.reject()

        assembly = vobj.Object.InList[0]
        if UtilsAssembly.activeAssembly() != assembly:
            Gui.ActiveDocument.setEdit(assembly)

        panel = TaskAssemblyCreateView(vobj.Object)
        Gui.Control.showDialog(panel)

######### Create Exploded View Task ###########
class TaskAssemblyCreateView(QtCore.QObject):
    def __init__(self, viewObj=None):
        super().__init__()

        view = Gui.activeDocument().activeView()

        self.assembly = UtilsAssembly.activeAssembly()
        self.assembly.ViewObject.EnableMovement = False
        self.asmDragger = self.assembly.ViewObject.getDragger()
        self.cbFin = view.addDraggerCallback(
            self.asmDragger, "addFinishCallback", self.draggerFinished
        )
        self.cbMov = view.addDraggerCallback(
            self.asmDragger, "addMotionCallback", self.draggerMoved
        )

        Gui.Selection.clearSelection()

        self.form = Gui.PySideUic.loadUi(":/panels/TaskAssemblyCreateView.ui")
        self.form.stepList.installEventFilter(self)
        self.form.stepList.itemClicked.connect(self.onItemClicked)

        self.saveAssemblyPartsPlacements(self.assembly)

        if viewObj:
            App.setActiveTransaction("Edit Exploded View")
            self.viewObj = viewObj
            for step in self.viewObj.Steps:
                step.Visibility = True
            self.onStepsChanged()

        else:
            App.setActiveTransaction("Create Exploded View")
            self.createExplodedViewObject()

        Gui.Selection.addSelectionGate(
            ExplodedViewSelGate(self.assembly, self.viewObj), Gui.Selection.ResolveMode.NoResolve
        )
        Gui.Selection.addObserver(self, Gui.Selection.ResolveMode.NoResolve)

        self.viewObj.Proxy.setStepsChangedCallback(self.onStepsChanged)
        self.callbackMove = view.addEventCallback("SoLocation2Event", self.moveMouse)
        self.callbackClick = view.addEventCallback("SoMouseButtonEvent", self.clickMouse)
        self.callbackKey = view.addEventCallback("SoKeyboardEvent", self.KeyboardEvent)

        self.selectingFeature = False
        self.form.LabelAlignDragger.setVisible(False)
        self.preselection_dict = None

        self.blockSetDragger = False
        self.blockDraggerMove = True
        self.currentStep = None

    def accept(self):
        self.deactivate()
        self.restoreAssemblyPartsPlacements(self.assembly)
        for step in self.viewObj.Steps:
            step.Visibility = False
        App.closeActiveTransaction()
        return True

    def reject(self):
        self.deactivate()
        App.closeActiveTransaction(True)
        return True

    def deactivate(self):
        pref = Preferences.preferences()
        pref.SetBool("PartsAsSingleSolid", self.form.CheckBox_PartsAsSingleSolid.isChecked())

        view = Gui.activeDocument().activeView()
        view.removeDraggerCallback(self.asmDragger, "addFinishCallback", self.cbFin)
        view.removeDraggerCallback(self.asmDragger, "addMotionCallback", self.cbMov)

        self.assembly.ViewObject.DraggerVisibility = False
        self.assembly.ViewObject.EnableMovement = True

        Gui.Selection.removeSelectionGate()
        Gui.Selection.removeObserver(self)
        Gui.Selection.clearSelection()

        self.viewObj.Proxy.setStepsChangedCallback(None)
        view.removeEventCallback("SoLocation2Event", self.callbackMove)
        view.removeEventCallback("SoMouseButtonEvent", self.callbackClick)
        view.removeEventCallback("SoKeyboardEvent", self.callbackKey)

        if Gui.Control.activeDialog():
            Gui.Control.closeDialog()
```

### Full version info

```shell
0.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-14:
@PaddleStroke off topic: nice visualization of the exploded view! Would it be possible to draw the dotted line from the joint position instead of the origin? Most imported STEPs and other assemblies are not drawn with a centered origin and then the lines would be way off in empty space.

PaddleStroke, 2024-03-15:
> All what's needed is to change your doubleClicked() method to return True.
> This tells the calling instance -- the tree view -- that the view provider
> handled the event.
> At the moment it doesn't have an explicit return value which is like None
> and this indicates the calling instance that the event wasn't handled.
> That's why two transactions are opened -- one for editing the joint and one
> for trying to rename the object.
> So, returning True fixes both of your problems.
> Btw, when I edit a joint then the title of the task panel still says

Thanks very much!!

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
closed 2024-05-06 by yorikvanhavre