FreeCAD/FreeCAD #12959: Presence of obsolete Assembly2 addon on local machine will break Addon Manager

luzpaz, 2024-03-14:
OK, just had the same issue in the latest stable 0.19 release. Running freecad from a terminal revealed an error in one of the workbench files.

This is the error.:
During initialization the error "invalid syntax (assembly2lib.py, line 59)" occurred in /home/tony/.FreeCAD Mod/FreeCAD_assembly2/InitGui.py

Deleting the offending folder (/home/tony/.FreeCAD Mod/FreeCAD_assembly2) solved the issue. It was probably a leftover from 0.18, but Addon manager really should be able to step over issues like this and just flag the workbench as faulty.

_Originally posted by @ozzyman in https://github.com/FreeCAD/FreeCAD-addons/issues/176#issuecomment-851201668_

chennes, 2024-03-14:
I cannot reproduce this error in current FreeCAD -- there is indeed a try/except block around the calls to load Init.py and InitGui.py, and they correctly handle the failed initialization. That error is report-view only, so we could consider popping a dialog if the Gui is up. Future work could include that dialog offering to rollback a recent Addon update (if there was one), or uninstalling/disabling the offending addon.

maxwxyz, 2024-03-16:
@luzpaz can you still reproduce this error? If not, either rephrase to be a FR for the mentioned rollback/disabling or we can close this issue.

JohnOCFII, 2024-10-25:
@luzpaz - can this be closed?

closed 2024-10-25 by luzpaz